Engineering-Research-and-Development/iotagent-opcua #6: [MUST] run tests on CI

jason-fox, 2019-01-18:
There are currently no CI Tests run on push. 
There is also no badge displaying the current test state on Travis (or equivalent)

```md
<br/>
[![Documentation badge](https://img.shields.io/readthedocs/iotagent-opcua.svg)](https://iotagent-opcua.rtfd.io/)
[![Build badge](https://img.shields.io/travis/Engineering-Research-and-Development/iotagent-opcua.svg)](https://travis-ci.org/Engineering-Research-and-Development/iotagent-opcua/)
![Status](https://nexus.lab.fiware.org/static/badges/statuses/iot-openmtc.svg)
```

You can look in the [IoTagent-LoRaWAN](https://github.com/Atos-Research-and-Innovation/IoTagent-LoRaWAN/blob/master/.travis.yml) to see how to set up the `.travis.yml` file.

- this is a TSC MUST requirement

jason-fox, 2019-01-21:
Downgraded from MUST to SHOULD since CI is running correctly and a stub exists. #9 

* running tests on CI is a TSC SHOULD requirement

gabrieledeluca, 2019-02-07:
Basic tests written

closed 2019-02-07 by gabrieledeluca