Engineering-Research-and-Development/iotagent-opcua #8: [MUST] Every GE must have training materials

jason-fox, 2019-01-18:
At the moment this enabler has no Tutorial and no link to the Academy and does not fulfil the minimum **Training** section requirements. 

The suggestion is to create a simple *Attaching other IoT Agents tutorial*  to provide a series of simple hello world connections for each agent in turn - this needs to include 

* One docker compose file for base config
* A set of REST API calls to provision the service
* An indication on how to make  minimal measure arrive (e.g. publishing an MQTT topic or equivalent)
* Proof that the data arrives in Orion.

- Presence of training  is a MUST requirement

gabrieledeluca, 2019-01-23:
Hello @jason-fox, what do you say about [https://iotagent-opcua.readthedocs.io/en/latest/opc_ua_agent_tutorial/index.html](https://iotagent-opcua.readthedocs.io/en/latest/opc_ua_agent_tutorial/index.html)? Thanks

jason-fox, 2020-01-29:
👍 LGTM. Closing

closed 2020-01-29 by jason-fox