Engineering-Research-and-Development/iotagent-opcua #20: Error launching docker image: "Config is not defined"

adr-arroyo, 2019-06-04:
Hello,

I am launching the agent in a docker-compose file along with some other FIWARE components and I am getting this error when the OPC UA agent launches:

```
iotage_1          | time=2019-06-04T11:16:53.891Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.Initialize | comp=iotAgent-OPCUA | srv=n/a | subsrv=n/a | msg=----------------------------------------------------
iotage_1          | time=2019-06-04T11:16:53.893Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.Initialize | comp=iotAgent-OPCUA | srv=n/a | subsrv=n/a | msg=endpointUrl         =  opc.tcp://<ipaddr>:<port>
iotage_1          | time=2019-06-04T11:16:53.893Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.Initialize | comp=iotAgent-OPCUA | srv=n/a | subsrv=n/a | msg=securityMode        =  NONE
iotage_1          | time=2019-06-04T11:16:53.895Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.Initialize | comp=iotAgent-OPCUA | srv=n/a | subsrv=n/a | msg=securityPolicy      =  0
iotage_1          | time=2019-06-04T11:16:53.897Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.Initialize | comp=iotAgent-OPCUA | srv=n/a | subsrv=n/a | msg=timeout             =  -1
iotage_1          | {"op":"IoTAgentNGSI.DomainControl","time":"2019-06-04T11:16:53.901Z","lvl":"ERROR","msg":"{ ReferenceError: config is not defined\n    
at Object.run (/opt/iotagent-opcua/iot_agent_modules/run/run.js:95:36)\n    
at /opt/iotagent-opcua/index.js:89:21\n    
at ChildProcess.exithandler (child_process.js:288:5)\n   
at emitTwo (events.js:126:13)\n    
at ChildProcess.emit (events.js:214:7)\n     
at maybeClose (internal/child_process.js:915:16)\n     
at Socket.stream.socket.on (internal/child_process.js:336:11)\n     
at emitOne (events.js:116:13)\n    at Socket.emit (events.js:211:7)\n     
at Pipe._handle.close [as _onclose] (net.js:561:12)\n  domain: \n   Domain {\n     domain: null,\n 
_events: { error: [Function: domainErrorHandler] },\n     _eventsCount: 1,\n     
_maxListeners: undefined,\n     members: [],\n      
start: 1559647009724,\n     trans: '6cab4c02-8950-4627-aa7d-84517463d3a3',\n     corr: '6cab4c02-8950-4627-aa7d-84517463d3a3',\n     op: 'IoTAgentNGSI.DeviceService' },\n  domainThrown: true }"}
```

My docker-compose configuration for the agent is:

```
  iotage:
    hostname: iotage
    image: rdlabengpa/opcuaage:latest
    networks:
      fiwareNetwork:
        aliases:
          - iotagent-opcua
    ports:
      - "4001:4001"
      - "4081:8080"
    depends_on:
      - mongodb
    volumes:
      - ./AGECONF:/opt/iotagent-opcua/conf
```

And the config.properties file looks like:

```
namespace-ignore=2
context-broker-host=orion
context-broker-port=1026
server-base-root=/
server-port=4001
device-registry-type=memory
mongodb-host=mongodb
mongodb-port=27017
mongodb-db=iotagent
mongodb-retries=5
mongodb-retry-time=5
fiware-service=<service>
fiware-service-path=<service-path>
provider-url=http://iotage:4001
device-registration-duration=P1M
endpoint=opc.tcp://<ipaddr>:<port>
log-level=DEBUG

#DATATYPE MAPPING OPCUA --> NGSI
OPC-datatype-Number=Number
OPC-datatype-Decimal128=Number
OPC-datatype-Double=Number
OPC-datatype-Float=Number
OPC-datatype-Integer=Integer
OPC-datatype-UInteger=Integer
OPC-datatype-String=Text
OPC-datatype-ByteString=Text
#END DATATYPE MAPPING OPCUA --> NGSI

#SESSION PARAMETERS
requestedPublishingInterval=500
requestedLifetimeCount=1000
requestedMaxKeepAliveCount=5
maxNotificationsPerPublish=10
publishingEnabled=true
priority=10

#SERVER CERT E AUTH
securityMode=NONE
securityPolicy=
userName=
password=

#Administration Services
api-port=4081
#End Administration Services

#POLL COMMANDS SETTINGS
polling=false
polling-commands-timer=30000
pollingDaemonFrequency=20000
pollingExpiration=200000
#END POLL COMMANDS SETTINGS

#AGENT ID
agent-id=iotagent-opcua

samplingInterval=1
queueSize=10
discardOldest=True
```

Actually the last parameters (samplingInterval, queueSize, discardOldest) are there because I was getting an error the first time I launched the agent saying that those were null, so I put them in the configuration file with some "random" values.

Are those errors due to not setting the fiware_path and fiware_service_path parameters? I am not sending any data to the agent yet.

Or because of the endpoint parameter? Actually this last parameter I do not know what it refers to, can I get some advice on what to put here?

Thanks and BR!

gabrieledeluca, 2019-06-04:
Hello,
the OPC UA Agent has a built-in Mapping Tool in order to create the (automatic) mapping between OPC UA nodes and NGSI entities/attributes. So, you should have an OPC UA Server (the _endpoint_ parameter of _configuration.properties_ file). Moreover, you need Orion Context Broker for communicating with the Northbound side (_context-broker-host_ and _context-broker-port_ parameters).

You could create the mapping file manually (or edit the Mapping Tool output) but you'll have other errors if you don't have both, the OPC UA Server and the OCB up and running.

BTW, we have implemented an OPC UA Server useful for testing purposes: [https://github.com/Engineering-Research-and-Development/opc-ua-car-server](https://github.com/Engineering-Research-and-Development/opc-ua-car-server)

Regarding monitoring parameters, I've added them also on the README page, thanks!

Please feel free to contact me if you need any further information.

Best Regards,
Gabriele

adr-arroyo, 2019-06-05:
Dear Gabriele,

Thanks for the fast response. I modified my configuration according to your answer. Now my files look like:

config.properties:

```
namespace-ignore=2
context-broker-host=orion
context-broker-port=1026
server-base-root=/
server-port=4001
device-registry-type=memory
mongodb-host=mongodb
mongodb-port=27017
mongodb-db=iotagentopc
mongodb-retries=5
mongodb-retry-time=5
fiware-service=UA
fiware-service-path=CarServer
provider-url=http://iotage:4001
device-registration-duration=P1M
endpoint=opc.tcp://aieng:5001
log-level=DEBUG

#DATATYPE MAPPING OPCUA --> NGSI
OPC-datatype-Number=Number
OPC-datatype-Decimal128=Number
OPC-datatype-Double=Number
OPC-datatype-Float=Number
OPC-datatype-Integer=Integer
OPC-datatype-UInteger=Integer
OPC-datatype-String=Text
OPC-datatype-ByteString=Text
#END DATATYPE MAPPING OPCUA --> NGSI

#SESSION PARAMETERS
requestedPublishingInterval=500
requestedLifetimeCount=1000
requestedMaxKeepAliveCount=5
maxNotificationsPerPublish=10
publishingEnabled=true
priority=10

#SERVER CERT E AUTH
securityMode=NONE
securityPolicy=
userName=
password=

#Administration Services
api-port=4081
#End Administration Services

#POLL COMMANDS SETTINGS
polling=false
polling-commands-timer=30000
pollingDaemonFrequency=20000
pollingExpiration=200000
#END POLL COMMANDS SETTINGS

#AGENT ID
agent-id=iotagent-opcua

#MONITORING PARAMETERS
samplingInterval= 0
queueSize= 10000
discardOldest= false
```

I have the OPC UA Car testing server running in opc.tcp://aieng:5001/UA/CarServer.

Additionally, my docker compose file for orion & the agent (I have more components but I do not place them all here):

```
  mongodb:
    image: mongo:3.2
    ports:
      - "27017:27017"
    expose:
      - "27017"
    command: --nojournal
    networks:
      - fiwareNetwork
    volumes:
      - "./data/db:/data/db"

  orion:
    image: fiware/orion:2.1.0
    ports:
      - "1026:1026"
    command:
      -dbhost mongodb
      -corsOrigin __ALL
      -logLevel DEBUG
    networks:
      - fiwareNetwork
    depends_on:
      - mongodb

  iotage:
    hostname: iotage
    image: rdlabengpa/opcuaage:latest
    networks:
      fiwareNetwork:
        aliases:
          - iotagent-opcua
    ports:
      - "4001:4001"
      - "4081:8080"
    depends_on:
      - mongodb
    volumes:
      - ./AGECONF:/opt/iotagent-opcua/conf
```

Despite this, I still have the same error in the docker-compose. Any clues on what could be happening??

BR,

Adrian

gabrieledeluca, 2019-06-05:
Dear Adrian,
the endpoint parameter has to contain the (complete) OPC UA Server URI, so in your case please set: 
`endpoint=opc.tcp://aieng:5001/UA/CarServer`

BR,
Gabriele

adr-arroyo, 2019-06-17:
Hello @gabrieledeluca ,
I continue here our conversation.

I was finally able to launch the agent, although it seems that it is required that the car-server is running before the agent starts or it (the agent) ends up in an error, am I correct?

However i am now receiving these error messages from the agent:

```
iotagent-opcua_1  | time=2019-06-17T13:36:08.974Z | lvl=ERROR | corr=6d078d4f-797c-4d2b-8d1b-845eb26ce28a | trans=6d078d4f-797c-4d2b-8d1b-845eb26ce28a | op=IoTAgentNGSI.DeviceService | srv=n/a | subsrv=n/a | msg=Registration error connecting to the Context Broker: {"code":"400","reasonPhrase":"Bad Request","details":"Only /absolute/ Service Paths allowed [a service path must begin with /]"} | comp=IoTAgent
iotagent-opcua_1  | time=2019-06-17T13:36:08.980Z | lvl=ERROR | corr=n/a | trans=n/a | op=Index.Initialize | comp=iotAgent-OPCUA | srv=UA | subsrv=CarServer | msg=could not register OCB context iotagent-opcuaCar
iotagent-opcua_1  | time=2019-06-17T13:36:08.982Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.Initialize | comp=iotAgent-OPCUA | srv=UA | subsrv=CarServer | msg={"name":"BAD_REQUEST","message":"Request error connecting to the Context Broker: {\"code\":\"400\",\"reasonPhrase\":\"Bad Request\",\"details\":\"Only /absolute/ Service Paths allowed [a service path must begin with /]\"}","code":400}
iotagent-opcua_1  | time=2019-06-17T13:36:08.988Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.InitSubscriptions | comp=iotAgent-OPCUA | srv=UA | subsrv=CarServer | msg=initializing monitoring: ns=1;s=Speed
iotagent-opcua_1  | time=2019-06-17T13:36:08.996Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.InitSubscriptions | comp=iotAgent-OPCUA | srv=UA | subsrv=CarServer | msg=initializing monitoring: ns=1;s=Acceleration
iotagent-opcua_1  | time=2019-06-17T13:36:09.002Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.InitSubscriptions | comp=iotAgent-OPCUA | srv=UA | subsrv=CarServer | msg=initializing monitoring: ns=1;s=EngineStatus
iotagent-opcua_1  | time=2019-06-17T13:36:09.003Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.InitSubscriptions | comp=iotAgent-OPCUA | srv=UA | subsrv=CarServer | msg=initializing monitoring: ns=1;s=Temperature
iotagent-opcua_1  | time=2019-06-17T13:36:09.006Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.InitSubscriptions | comp=iotAgent-OPCUA | srv=UA | subsrv=CarServer | msg=initializing monitoring: ns=1;s=Oxigen
iotagent-opcua_1  | time=2019-06-17T13:36:09.080Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.InitSubscriptions | comp=iotAgent-OPCUA | srv=UA | subsrv=CarServer | msg=started subscription:  613588
iotagent-opcua_1  | time=2019-06-17T13:36:09.088Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.InitSubscriptions | comp=iotAgent-OPCUA | srv=UA | subsrv=CarServer | msg= revised parameters 
iotagent-opcua_1  | time=2019-06-17T13:36:09.089Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.InitSubscriptions | comp=iotAgent-OPCUA | srv=UA | subsrv=CarServer | msg=  revised maxKeepAliveCount   5  ( requested  5)
iotagent-opcua_1  | time=2019-06-17T13:36:09.091Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.InitSubscriptions | comp=iotAgent-OPCUA | srv=UA | subsrv=CarServer | msg=  revised lifetimeCount       1000  ( requested  1000)
iotagent-opcua_1  | time=2019-06-17T13:36:09.094Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.InitSubscriptions | comp=iotAgent-OPCUA | srv=UA | subsrv=CarServer | msg=  revised publishingInterval  500  ( requested  500)
iotagent-opcua_1  | time=2019-06-17T13:36:09.095Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.InitSubscriptions | comp=iotAgent-OPCUA | srv=UA | subsrv=CarServer | msg=  suggested timeout hint      10000
iotagent-opcua_1  | time=2019-06-17T13:36:09.101Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.InitSubscriptions | comp=iotAgent-OPCUA | srv=UA | subsrv=CarServer | msg=started subscription:  613589
iotagent-opcua_1  | time=2019-06-17T13:36:09.107Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.InitSubscriptions | comp=iotAgent-OPCUA | srv=UA | subsrv=CarServer | msg= revised parameters 
iotagent-opcua_1  | time=2019-06-17T13:36:09.108Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.InitSubscriptions | comp=iotAgent-OPCUA | srv=UA | subsrv=CarServer | msg=  revised maxKeepAliveCount   5  ( requested  5)
iotagent-opcua_1  | time=2019-06-17T13:36:09.110Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.InitSubscriptions | comp=iotAgent-OPCUA | srv=UA | subsrv=CarServer | msg=  revised lifetimeCount       1000  ( requested  1000)
iotagent-opcua_1  | time=2019-06-17T13:36:09.111Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.InitSubscriptions | comp=iotAgent-OPCUA | srv=UA | subsrv=CarServer | msg=  revised publishingInterval  500  ( requested  500)
iotagent-opcua_1  | time=2019-06-17T13:36:09.113Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.InitSubscriptions | comp=iotAgent-OPCUA | srv=UA | subsrv=CarServer | msg=  suggested timeout hint      10000
iotagent-opcua_1  | time=2019-06-17T13:36:09.132Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.InitSubscriptions | comp=iotAgent-OPCUA | srv=UA | subsrv=CarServer | msg=started subscription:  613590
iotagent-opcua_1  | time=2019-06-17T13:36:09.133Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.InitSubscriptions | comp=iotAgent-OPCUA | srv=UA | subsrv=CarServer | msg= revised parameters 
iotagent-opcua_1  | time=2019-06-17T13:36:09.134Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.InitSubscriptions | comp=iotAgent-OPCUA | srv=UA | subsrv=CarServer | msg=  revised maxKeepAliveCount   5  ( r [... cut ...] ,"code":404}
```

After a few of those, I am getting these kind of messages: 

[... 165 lines cut ...]
securityPolicy=
userName=
password=

#Administration Services
api-port=4081
#End Administration Services

#POLL COMMANDS SETTINGS
polling=false
polling-commands-timer=30000
pollingDaemonFrequency=20000
pollingExpiration=200000
#END POLL COMMANDS SETTINGS

#AGENT ID
agent-id=iotagent-opcua

#MONITORING PARAMETERS
samplingInterval= 0
queueSize= 10000
discardOldest= false
```

Thanks for the help!

gabrieledeluca, 2019-06-17:
Dear Adrian,
yes the OPC UA server has to be up and running before launching the agent, for this reason, you can set this precondition (depends_on clause) on docker-compose. 

Regarding errors, it seems that there is a communication error with the OCB, please could you verify the OCB status? Moreover, I see some differences between the config.properties and the config.json file (I'm referring both FIWARE service and service-path attributes), have you edited the config.json manually?

Best Regards,
Gabriele

adr-arroyo, 2019-06-18:
Dear gabriele,

OCB is ok as far as I understand. I editted config.properties with that configuration (fiware & fiware service-path) but I did not change config.json manually. 

I believe the problem is coming from the endpoint, as I got this error:

```
iotagent-opcua_1  | time=2019-06-18T07:41:54.698Z | lvl=ERROR | corr=n/a | trans=n/a | op=Index.Initialize | comp=iotAgent-OPCUA | srv=n/a | subsrv=n/a | msg= error { Error: getaddrinfo ENOTFOUND aieng aieng:5001
iotagent-opcua_1  |     at GetAddrInfoReqWrap.onlookup [as oncomplete] (dns.js:67:26)
iotagent-opcua_1  |   errno: 'ENOTFOUND',
iotagent-opcua_1  |   code: 'ENOTFOUND',
iotagent-opcua_1  |   syscall: 'getaddrinfo',
iotagent-opcua_1  |   hostname: 'aieng',
iotagent-opcua_1  |   host: 'aieng',
iotagent-opcua_1  |   port: 5001 }
iotagent-opcua_1  | time=2019-06-18T07:41:54.701Z | lvl=ERROR | corr=n/a | trans=n/a | op=Index.Initialize | comp=iotAgent-OPCUA | srv=n/a | subsrv=n/a | msg= stack trace Error: getaddrinfo ENOTFOUND aieng aieng:5001`
```

What do you think?

BR,
Adrian

PietroGreco, 2019-06-24:
Hi Adrian,
I have just uploaded a working full test bed example:
https://github.com/Engineering-Research-and-Development/opc-ua-car-server/blob/master/FullTestBed.tar.gz
1) mkdir fiware-testbed
2) cd fiware-testbed
3) Extract FullTestBed.tar.gz within this folder
4) docker-compose up

Once you have launched the services you can visit http://localhost. You will find a dashboard to control the CarServer.

Inside _fiware-testbed_ folder you can find an example of _docker-compose.yml_ and _config.properties_
Be aware that if you change fiware-service and fiware-service-path in config.properties you also have to update fiware-service and fiware-servicepath for all the *.sh files inside CAR_CLIENT folder

This testbed is a good starting point.

closed 2019-11-06 by gabrieledeluca