Engineering-Research-and-Development/iotagent-opcua #29: Variables not being updated in OCB

gabrielmbmb, 2019-10-09:
Hey guys, I created an OPC UA server following this tutorial https://github.com/node-opcua/node-opcua/blob/master/documentation/creating_a_server.md. I'm using node-opcua v2.1.9. The server is running fine, but the IoT agent is only getting values from it when it is started. Then, when the OPCUA server updates the value of a variable, the agent isn't realizing it and it isn't getting the new variable value.  I also tested my server with this tool https://github.com/node-opcua/opcua-commander and it got the value of the variables when they were updated.

I've tested the IoT Agent with your OPC UA Server example https://github.com/Engineering-Research-and-Development/opc-ua-car-server and it worked fine.

I've been stuck with this problem now almost 2 weeks. When I first started, I've seen that the agent was using the node-opcua v0.0.61. Now I've seen that you are updating it to node-opcua v2.1.8, but the docker image seems to be using the old version yet. 

Could it be a problem between node-opcua version in client and server?

PietroGreco, 2019-10-10:
Hi Gabriel,

Yes, I think the issue is related to the two different versions.
Anyway, yesterday we pre-released the latest OPC UA IotAgent version which uses node-opcua 2.1.8. We have followed a docker first approach so the images are up-to-date too.

I have also created a very simple OPC UA server following that tutorial and tested the communication between it and the Agent. Everything now works fine.

Please download the latest Agent version and follow these [instructions](https://github.com/Engineering-Research-and-Development/iotagent-opcua/blob/master/docs/user_and_programmers_manual.md#step-2---map-ip-addresses)

You can start from Step 2

**NOTE:** Be aware that I had to add this snippet of code to make the server accept unknown certificates.
```    
serverCertificateManager: new opcua.OPCUACertificateManager({ 
        automaticallyAcceptUnknownCertificate: true
})
```
I enclose the complete sample server code:
```
/*global require,setInterval,console */
const opcua = require("node-opcua");

// Let's create an instance of OPCUAServer
const server = new opcua.OPCUAServer({
    port: 5001, // the port of the listening socket of the server
    resourcePath: "/UA/MyLittleServer", // this path will be added to the endpoint resource name
    buildInfo : {
      productName: "MySampleServer1",
      buildNumber: "7658",
      buildDate: new Date(2014,5,2)
    },
    serverCertificateManager: new opcua.OPCUACertificateManager({ 
        automaticallyAcceptUnknownCertificate: true,
        // rootFolder: path.join(__dirname, "../certs")
    })
});

function post_initialize() {
    console.log("initialized");

    function construct_my_address_space(server) {

      const addressSpace = server.engine.addressSpace;
      const namespace = addressSpace.getOwnNamespace();

      // declare a new object
      const device = namespace.addObject({
         organizedBy: addressSpace.rootFolder.objects,
         browseName: "MyDevice"
      });

      // add some variables
      // add a variable named MyVariable1 to the newly created folder "MyDevice"
        let variable1 = 1;

        // emulate variable1 changing every 500 ms
        setInterval(function(){  variable1+=1; }, 500);

        namespace.addVariable({
            componentOf: device,
            browseName: "MyVariable1",
            dataType: "Double",
            value: {
                get: function () {
                    return new opcua.Variant({dataType: opcua.DataType.Double, value: variable1 });
                }
            }
        });
  }

  construct_my_address_space(server);
  server.start(function() {
        console.log("Server is now listening ... ( press CTRL+C to stop)");
        console.log("port ", server.endpoints[0].port);
        const endpointUrl = server.endpoints[0].endpointDescriptions()[0].endpointUrl;
        console.log(" the primary server endpoint url is ", endpointUrl );
    });
}
server.initialize(post_initialize);
```

Let me know and feel free to contact us for any questions

gabrielmbmb, 2019-10-15:
Hi guys,

I used the latest Docker image and now the variables are being updated by the IoT Agent correctly!

Thank you so much.

closed 2019-10-15 by gabrielmbmb