Engineering-Research-and-Development/iotagent-opcua #30: Modify OPC-UA variables from entity property update on Context Broker

ismaeldols, 2019-10-10:
Hello,

I would like to write on an OPC-UA server variable directly when the context broker receives an entity property update request. Is this possible with this agent? If so, how could I do this?

I have read your tutorial and I can't find this anywhere. The only way I have found to do this is using an OPC-UA command.

Thanks.

ismaeldols, 2019-10-18:
Please, I need to know if this functionality is available.
Thanks

PietroGreco, 2019-10-18:
Hi Ismael,
That functionality is not available

ismaeldols, 2019-10-18:
Thanks, Do you plan to implement this feature in the future?

PietroGreco, 2019-10-18:
At the moment we do not have a plan for the implementation of this feature.

ismaeldols, 2019-10-23:
Ok, Thanks

ismaeldols, 2019-11-06:
I woud really appreciate it if you could help me with this point, thank you.

gabrieledeluca, 2019-11-06:
Dear Ismael, 
firstly we should analyze how we could integrate functionality you mentioned keeping the current workflow based on the IDAS framework (let me say that cannot be planned for the short-term). 
The IDAS framework supports communication to the physical level using commands, for that reason, the current implementation of the agent is not able to satisfy your needs. 
So, in order to reach your objective, you should write an external application with the logic for translating commands and send write requests to the OPC UA server (i.e., you should create a subscription to the OCB in order to receive update requests, code the business logic for exposing an endpoint, parsing the request - processing only requests coming from a context consumer, skipping updates coming from the agent representing the physical level to avoid loops - and finally sending requests to the OPC UA Server).

manfredipist, 2023-01-12:
Hi @ismaeldols , i just wanted to let you know that the feature, you've been requesting, has been finally implemented on latest version of the agent (v2.0.5)

closed 2023-01-12 by manfredipist