Engineering-Research-and-Development/iotagent-opcua #33: Registering variable with node_id with quotes

gabrielmbmb, 2020-01-23:
Hello,

I'm trying to register a variable with the following node_id in a OPC UA server: ```ns=3;s="input1"```. I'm sending the following JSON to the OPC UA Agent:

```json
{
    "devices": [
        {
            "device_id": "plc_demo",
            "entity_name": "urn:ngsi-ld:PLC:001",
            "entity_type": "Device",
            "attributes": [
                {
                    "object_id": "ns*3:s*\"input1\"",
                    "name": "input1",
                    "type": "Number" 
                }          
            ],
            "lazy": [],
            "commands": []
        }
    ]
}
```

I tried to escape the double quotes in the JSON but the agent responds "WRONG_SYNTAX". Also, I tried to use ```"ns*3:s*input1"```, but the agent didn't find the node in the address space (because the node_id is ```ns=3;s="input1"``` not ```ns=3;s=input1```) and ```"ns*3:s*\\\"input1\\\""``` but the agent responded WRONG_SYNTAX again. I couldn't find any info about this in the documentation. So, there is a way to register a variable with quotes in its node_id?

Thank you so much.

gabrielmbmb, 2020-01-23:
I was having a look to the iotagent-node-lib and seems like it is not a specif problem of your agent and instead of the iotagent-node-lib...

gabrieledeluca, 2020-01-29:
Hello @gabrielmbmb, I opened an issue about that [https://github.com/telefonicaid/iotagent-node-lib/issues/671](https://github.com/telefonicaid/iotagent-node-lib/issues/671) in order to relax constraints about the _object_id_ value (related to the Southbound part, so should be free) probably @jason-fox could give us news about that. 

In any case, we could provide you workaround (as we did for special chars, = and ;, usually contained in OPC UA node IDs) or you can edit by yourself the parsePayloadProperties function in the run.js file choosing, on you own, the binding you will use for "translating" the double-quotes char (choosing the right one to avoid potential conflicts with your OPC UA address space).

gabrielmbmb, 2020-01-29:
Yeah, I saw the issue a few days ago... I'll take your recommendation and I will modify the parsePayloadProperties on my own. Maybe is not a bad idea to have the replacement of double quotation marks in the repo until it is changed in https://github.com/telefonicaid/iotagent-node-lib If you don't mind I can create a PR. Maybe single quotation marks could be used to replace double quotes. What do you think?

gabrieledeluca, 2020-01-29:
OK for the PR, but using the single quotes you will have the same problems because currently, you cannot put in the object_id the Orion forbidden chars (please find [here ](https://fiware-orion.readthedocs.io/en/master/user/forbidden_characters/index.html)the list). Honestly, I don't like this approach because we will be in trouble if the OPC UA node ID will contain (the "original one" coming from the OPC UA server) one of the binding chars we chose for the workaround.

gabrielmbmb, 2020-01-29:
Yeah, I just realized that single quotes are forbidden, too. I also don't like this approach for the same reason... Maybe, instead of replacing double quotes with a single character we could use several characters or a very obvious word to avoid problems with the real node ID like "&dq" standing for double quotation mark.

gabrieledeluca, 2020-01-29:
OK, your approach could mitigate the problem but it is not a definitive solution, an OPC UA NodeId could contain &dq, opening an issue. There are no limitations about the value of the NodeID in the OPC UA spec (have a look to the OPC UA specifications [here](https://opcfoundation.org/developer-tools/specifications-unified-architecture/part-3-address-space-model/), par. 8.2).

Anyway, you can create the PR adding the workaround you suggested.

An alternative solution for you could be the customization of the iot-agent-lib dropping the validation of the object_id (as I did [here](https://github.com/gabrieledeluca/iotagent-node-lib/commit/508d89578e4ae7c445a69fdd5cb2ea0957b852ae)).

closed 2020-07-19 by gabrielmbmb