Engineering-Research-and-Development/iotagent-opcua #42: support of (metadata) source timestamps

joriwinderickx-fm, 2020-12-09:
I have a basic FIWARE architecture set up that has roughly the following dataflow:

> OPC-UA simulation server -> IoT Agent OPC-UA -> Orion -> QuantumLeap -> CrateDB

The configuration of the IoT agent was based on the examples that were available, though, the IoT Agent does not seem to take into account the timestamp (Source Timestamp) of the value of the node that is configured to be pushed to Orion. Is it supported by the IoT agent e.g. to include this timestamp as a metadata value? If yes, is it documented somewhere on how to configure it?

You can find the configuration that was used for the IoT agent attached:
[config.properties.txt](https://github.com/Engineering-Research-and-Development/iotagent-opcua/files/5665043/config.properties.txt)
[config.json.txt](https://github.com/Engineering-Research-and-Development/iotagent-opcua/files/5665044/config.json.txt)

rosieks, 2022-10-20:
I would be happy to have that working with NGSI-LD and having `observedAt` value present in the request

MattiaMarzano-Eng, 2022-10-20:
The support to metadata will be available in the newer version (v2) of the OPC UA Agent that will be released soon. See the dedicated [branch](https://github.com/Engineering-Research-and-Development/iotagent-opcua/tree/iotagent-v2)

MattiaMarzano-Eng, 2023-09-28:
Agent v2 available on the main branch

closed 2023-09-28 by MattiaMarzano-Eng