Engineering-Research-and-Development/iotagent-opcua #46: How to test iotagent opcua with dummy device data.

daminichopra, 2021-03-05:
Hi,

I want to test iotagent opcua with my dummy device data. I have followed the steps described in the below document.
https://github.com/Engineering-Research-and-Development/iotagent-opcua/blob/master/docs/opc_ua_agent_tutorial.md

Install iotagent-opcua using docker
Provision a new Device 
update from Orion.
Now, I want to check the flow is working correctly by sending the value from opcua to orion as to check End to end dummy data flow with Orion.
Also, I don't have the actual device/sensor. I want to send data using my dummy device.

Please guide me how I can test with my dummy device and how can service provisioning can be done using opcua?
Also, please confirm which payload format used by iotagent-opcua

daminichopra, 2021-03-08:
Gentle Reminder!!

manfredipist, 2021-03-08:
Hello @daminichopra , could you please be more specific? what kind of problems are you encountering? If you've followed the doc (https://github.com/Engineering-Research-and-Development/iotagent-opcua/blob/master/docs/opc_ua_agent_tutorial.md) you should have a fully working environment made of: a client (iotagent), a server (opcuacarsrv), a context broker (orion) and a storage solution (orion_mongo, iotmongo). Consequently, you should be able to interact with your server by sending commands to the context broker (orion)

daminichopra, 2021-03-08:
Hi @manfredipist , Thank you for the response.
I have successfully done the device provisioning and also opcua connected to orion context broker but further I want to send dummy payload from opcua to context broker- orion. So, is there any API supported or any other method by which it can be done. Also, I don't have actual sensor/device.

manfredipist, 2021-03-08:
I think you may have misunderstood the flow, it's the agent that writes data to the context broker either with on-demand requests (lazy attributes and commands) or by subscribing to a server (active attributes), not the server itself.

daminichopra, 2021-03-08:
Yes Right!! So how can I send dummy device payload from opcua agent to orion if I don't have actual sensor/device.

manfredipist, 2021-03-08:
Your dummy device should be a fully compliant opcua server and your specific attribute (temperature in this case) should be readable. You can check our implementation of an opcua server to build your own (https://github.com/Engineering-Research-and-Development/opc-ua-car-server). We don't actually support agent-orion API communication without a working server.

daminichopra, 2021-03-10:
> Hello @daminichopra , could you please be more specific? what kind of problems are you encountering? If you've followed the doc [...]

Hi @manfredipist , Actually i have done the deployment of opcua setup successfully. Now by following [doc](https://iotagent-opcua.readthedocs.io/en/latest/opc_ua_agent_tutorial/index.html)  i am able to province a new device and also able to execute all API given in doc . Next, I want to check Northbound as well as Southbound traffic end to end dummy data flow with orion(context broker) using dummy device as I don't have actual sensor/device.

gabrieledeluca, 2021-03-11:
Hello @daminichopra in order to test the OPC UA agent you need to have an OPC UA Server in place. It could be real or fake/simulated, but it is a mandatory component for testing. 
The OPC UA Agent creates an OPC UA session for monitoring variables, uses OPC UA read/write services for reading variables and executing methods. In a nutshell, it uses the OPC UA standards for everything. 
BTW, if you don't have a server, you can easily create a new one from scratch (using your favourite library) or you can deploy the (open source) server we created for testing purpose [Testing Section](https://github.com/Engineering-Research-and-Development/iotagent-opcua#testing).

daminichopra, 2021-03-12:
> Hello @daminichopra in order to test the OPC UA agent you need to have an OPC UA Server in place. It could be real or fake/simulated, but it is a mandatory component for testing.
> The OPC UA Agent [...]

Hi @gabrieledeluca , Thank you for response.

I am also trying to do setup of iotagent-opcua using docker and followed below steps for same:

1.   git clone "https://github.com/Engineering-Research-and-Development/iotagent-opcua" 
2.   Used [docker-compose-external-server.yml](https://github.com/Engineering-Research-and-Development/iotagent-opcua/blob/master/docker-compose-external-server.yml) file for agent and [docker-compose.yml](https://github.com/Engineering-Research-and-Development/iotagent-opcua/blob/master/docker-compose.yml) for carserver.
3. Next my opcua-agent for version 1.3.4 ...container is crashing down showing below logs:
    
[image]

For this issue i tried generating certificates client_certificate.pem and client_private_key.pem and added path in [docker-compose.yml. ](https://github.com/Engineering-Research-and-Development/iotagent-opcua/blob/6b67a59c9fba9e0203add0ff91d57e11bf02bfbe/docker-compose.yml#L29)
Also, if i use latest image of  iotagent4fiware/iotagent-opcua:latest docker container it shows no logs and by using Api -  `curl localhost:4001/version
curl: (56) Recv failure: Connection reset by peer
`

Please provide your support for issue which i have described above. Thank you

daminichopra, 2021-03-15:
> Hello @daminichopra in order to test the OPC UA agent you need to have an OPC UA Server in place. It could be real or fake/simulated, but it is a mandatory component for testing.
> The OPC UA Agent creates an OPC UA session for monitoring variables, uses OPC UA read/write services for reading variables and executing methods. In a nutshell, it uses the OPC UA standards for everything.
> BTW, if you don't have a server, you can easily create a new one from scratch (using your favorite library) or you can deploy the (open source) server we created for testing purpose [Testing Section](https://github.com/Engineering-Research-and-Development/iotagent-opcua#testing).

Hi @gabrieledeluca , I have successfully able to run OPC UA server and OPC UA agent . For further testing if I have to update value of car object attributes how can i do that using dummy payload to test Northbound traffic? . Also, i have  referred [presentation ](https://www.slideshare.net/FI-WARE/fiware-global-summit-developing-an-iot-agent-case-example-opc-ua-147767858) and in `slideno - 31 ` using `drop_db.sh` is dropping `database - orion-opcua_car`  from orion . Please give some overview on drop_db.sh used . Thank you

daminichopra, 2021-03-16:
Soft Reminder!!

Anupamskd7, 2022-08-17:
Hi @gabrieledeluca, according to the [doc](https://iotagent-opcua.readthedocs.io/en/latest/opc_ua_agent_tutorial/index.html), I have deployed OPC UA server, OPC UA Agent.
Now, following the doc, I provisioned the "car" device and updated the Orion entity for "accelerate" attribute.

What I want is to send device alerts (Northbound data) from IoT Agent which will update it on Orion CB. (Like send attribute to 7896 port in other IoTAgents).

Also, if this is done via the OPC UA Server, could you please share the procedure?

Anupamskd7, 2022-08-22:
Hi @gabrieledeluca @manfredipist, I would really appreciate if you could help me in this. Thanks

manfredipist, 2023-01-12:
@Anupamskd7  please download the latest version of the agent (v.2.0.5) and try the tutorial provided. If you still encounter problems, please open a new issue

closed 2023-01-12 by manfredipist