Engineering-Research-and-Development/iotagent-opcua #48: npm ERR

VANHUUTHANHTRUNG, 2021-05-17:


tstorek, 2021-05-27:
hello,

I have the same issue here. Additionally, I am very confused by the different docs. When I understand it correctly. The default ports in the DOCKERFILE also do not match the ones in the default docker-compose.yml. However, I think that the command is an artifact from an old version. Without the agent seems to start up and also response on the admin-endpoints with "running"

VANHUUTHANHTRUNG, 2021-05-28:
@tstorek  I have found work-around fixes you can try if they help:

- Reinstall npm
- Remove the `"start" : "node index.js"` from `scripts` part in `package.json` file
- In `docker-compose.yml` file, change `command: npm start` into `command: index.js`

I am able to run the tutorial but I have not gotten the cause of this ERROR. Please try and let me know what you would get.

tstorek, 2021-05-28:
hi,

I think that you do not even need the command. I ran it without. and it seemed to work. At least I could send and receive :) running index.js as command does not sound reasonable to me ;)