Engineering-Research-and-Development/iotagent-opcua #52: Can not access Lazy attribute from Orion

VANHUUTHANHTRUNG, 2021-06-22:
## Context ##
Used code base and tutorial: [FIWARE IoT Agent for OPC-UA tutorial][1] 

## Target ##

 - To access all available attributes both from IoT Agent and Orion Context Broker
 - To modify attributes via commands sent to Orion Context Broker then passed down to IoT Agent

## Replicate steps ##
### Clone repo and start containers ###
```
git clone "https://github.com/Engineering-Research-and-Development/iotagent-opcua"
cd iotagent-opcua
docker-compose up -d
#
```

I found this error from the log which related to DEVICE_GROUP_NOT_FOUND
```
time=2021-06-21T06:52:38.105Z | lvl=DEBUG | corr=n/a | trans=n/a | op=IoTAgentNGSI.MongoDBGroupRegister | srv=n/a | subsrv=n/a | msg=Device group for fields [["type"]] not found: [{"type":"Device"}]
time=2021-06-21T06:52:38.105Z | lvl=DEBUG | corr=n/a | trans=n/a | op=IoTAgentNGSI.NGSIService | srv=opcua_car | subsrv=/demo | msg=error {"name":"DEVICE_GROUP_NOT_FOUND","message":"Couldn\t find device group for fields: [\"type\"] and values: {\"type\":\"Device\"}","code":404} in get group device
```

### Provision device ###

```
curl http://localhost:4001/iot/devices \
     -H "fiware-service: opcua_car" \
     -H "fiware-servicepath: /demo" \
     -H "Content-Type: application/json" \
     -d @./testCommands/add_device.json
```
Successfully provision new device

### Send accelerate commands to both entities ###
```
curl -X PUT \
  'http://localhost:1026/v2/entities/age01_Car/attrs/Accelerate?type=Device' \
  -H 'content-type: application/json' \
  -H 'fiware-service: opcua_car' \
  -H 'fiware-servicepath: /demo' \
  -d '{
  "value": [2],
  "type": "command"
}'

curl -X PUT \
  'http://localhost:1026/v2/entities/age02_Car/attrs/Accelerate?type=Device' \
  -H 'content-type: application/json' \
  -H 'fiware-service: opcua_car' \
  -H 'fiware-servicepath: /demo' \
  -d '{
  "value": [2],
  "type": "command"
}'
```
**Logs from Orion**
```
INFO@07:08:37  logTracing.cpp[130]: Request received: PUT /v2/entities/age01_Car/attrs/Accelerate?type=Device, request payload (39 bytes): {
  "value": [2],
  "type": "command"
}, response code: 204

```
**Logs from Agent**
```
time=2021-06-21T07:12:49.845Z | lvl=DEBUG | corr=n/a | trans=n/a | op=IoTAgentNGSI.GenericMiddlewares | msg=Request for path [//op/update] from [iotage:4001]
time=2021-06-21T07:12:49.845Z | lvl=DEBUG | corr=n/a | trans=n/a | op=IoTAgentNGSI.GenericMiddlewares | msg=Body:

{
[... 222 lines cut ...]
    volumes:
      - orion_mongo_data:/data/db
      - orion_mongo_conf:/data/configdb
    command: --nojournal # add a buffer for changes made in mongodb with the cost of doubling the memory

volumes:
  iot_mongo_data:
  iot_mongo_conf:
  orion_mongo_data:
  orion_mongo_conf:

networks:
  hostnet:
  iotnet:
  ocbnet:

```

## Issue ##
 - Not allow to retrieve lazy attribute as in the tutorial
 - Unknown Error in IoT Agent's log

I can however access other attributes.

  [1]: https://github.com/Engineering-Research-and-Development/iotagent-opcua/blob/master/docs/opc_ua_agent_tutorial.md

manfredipist, 2021-06-30:
Hello @VANHUUTHANHTRUNG , we may  have finally found the issue, you should remove "ngsiVersion" from your config.json while using orion (not ld one). Please let us know if it fixes your issue

VANHUUTHANHTRUNG, 2021-07-05:
# Update on recommended fix #

@manfredipist If we remove `ngsiVersion`, will ngsiV2 be used instead ?

# Steps to replicate #
- Clone project
-  Remove both `ngsiVersion` (I also tried with and without `jsonLdContext`)
- Run curl commands from the tutorial to test.
Here is 2 `config.json` in both `AGECONF` and `conf` dir:
[image]
- run `docker-compose up -d`
*The `config.json` file used in both directories*
```
{
	"logLevel": "DEBUG",
	"multiCore": false,
	"relaxTemplateValidation": true,
	"contextBroker": {
		"host": "orion",
		"port": 1026,
		"service": "opcua_car",
		"subservice": "/demo"
	},
	"server": {
		"port": 4001,
		"baseRoot": "/"
	},
	"deviceRegistry": {
		"type": "memory"
	},
	"mongodb": {
		"host": "iot_mongo",
		"port": "27017",
		"db": "iotagent",
		"retries": 5,
		"retryTime": 5
	},
	"types": {
		"Device": {
			"service": "opcua_car",
			"subservice": "/demo",
			"active": [{
				"name": "EngineBrake",
				"type": "Number"
			}, {
				"name": "Acceleration",
				"type": "Number"
			}, {
				"name": "EngineStopped",
				"type": "Boolean"
			}, {
				"name": "Engine_Temperature",
				"type": "Number"
			}, {
				"name": "Engine_Oxigen",
				"type": "Number"
			}],
			"lazy": [{
				"name": "Speed",
				"type": "Number"
			}],
			"commands": [{
				"name": "Error",
				"type": "command"
			}, {
				"name": "Stop",
				"type": "command"
			}, {
				"name": "Accelerate",
				"type": "command"
[... 304 lines cut ...]
"value":{
"@type":"commandStatus",
"@value":"UNKNOWN"
},
"mdNames":[
]
},
"https://uri=etsi=org/ngsi-ld/default-context/ToggleSensorActivation_info":{
"type":"Property",
"creDate":1625477977.7794368,
"modDate":1625477977.7794368,
"value":{
"@type":"commandResult",
"@value":" "
},
"mdNames":[
]
}
},
"creDate":1625477977.7794368,
"modDate":1625477978.4732308,
"lastCorrelator":""
}
```
Thank you for your support.

manolait, 2021-10-04:
Hello,

I have exactly the same problem. I follow the tutorial step by step and no entity is created in OCB.
```

manuelmunozrodriguez@MacBook-Pro iotagent-opcua % curl -X GET \
  http://localhost:1026/v2/entities/age01_Car/attrs/Speed \
  -H 'fiware-service: opcua_car' \
  -H 'fiware-servicepath: /demo'
{"error":"NotFound","description":"The requested entity has not been found. Check type and id"}%                                                                                                          manuelmunozrodriguez@MacBook-Pro iotagent-opcua % curl http://localhost:4001/iot/devices \
     -H "fiware-service: opcua_car" \
     -H "fiware-servicepath: /demo" \
     -H "Content-Type: application/json" \
     -d @./testCommands/add_device.json
{}%                                                                                                                                                                                                       manuelmunozrodriguez@MacBook-Pro iotagent-opcua % curl http://localhost:4001/iot/devices \
     -H "fiware-service: opcua_car" \
     -H "fiware-servicepath: /demo" \
     -H "Content-Type: application/json" \
     -d @./testCommands/add_device.json
{"name":"DUPLICATE_DEVICE_ID","message":"A device with the same pair (Service, DeviceId) was found:age02_Car"}%                                                                                           manuelmunozrodriguez@MacBook-Pro iotagent-opcua % curl -X PUT \
  'http://localhost:1026/v2/entities/age01_Car/attrs/Accelerate?type=Device' \
  -H 'content-type: application/json' \
  -H 'fiware-service: opcua_car' \
  -H 'fiware-servicepath: /demo' \
  -d '{
  "value": [2],
  "type": "command"
}'

curl -X PUT \
  'http://localhost:1026/v2/entities/age02_Car/attrs/Accelerate?type=Device' \
  -H 'content-type: application/json' \
  -H 'fiware-service: opcua_car' \
  -H 'fiware-servicepath: /demo' \
  -d '{
  "value": [2],
  "type": "command"
}'
{"error":"NotFound","description":"The requested entity has not been found. Check type and id"}{"error":"NotFound","description":"The requested entity has not been found. Check type and id"}% 
```

What is the problem? How can I solve it?

manfredipist, 2021-10-04:
hello @manolait , please show us your config.json, config.properties and your docker-compose file

manolait, 2021-10-04:
Hello,

Thank you for your promptness. I am launching everything just like the documentation

File /config/config.json

```
{
	"logLevel": "DEBUG",
	"multiCore": false,
	"relaxTemplateValidation": true,
	"contextBroker": {
		"host": "orion",
		"port": 1026,
		"ngsiVersion": "ld",
		"jsonLdContext": "https://uri.etsi.org/ngsi-ld/v1/ngsi-ld-core-context.jsonld",
		"service": "opcua_car",
		"subservice": "/demo"
	},
	"server": {
		"port": 4001,
		"baseRoot": "/"
	},
	"deviceRegistry": {
		"type": "memory"
	},
	"mongodb": {
		"host": "iot_mongo",
		"port": "27017",
		"db": "iotagent",
		"retries": 5,
		"retryTime": 5
	},
	"types": {
		"Device": {
			"service": "opcua_car",
			"subservice": "/demo",
			"active": [{
				"name": "EngineBrake",
				"type": "Number"
			}, {
				"name": "Acceleration",
				"type": "Number"
			}, {
				"name": "EngineStopped",
				"type": "Boolean"
			}, {
				"name": "Engine_Temperature",
				"type": "Number"
			}, {
				"name": "Engine_Oxigen",
				"type": "Number"
			}],
			"lazy": [{
				"name": "Speed",
				"type": "Number"
			}],
			"commands": [{
				"name": "Error",
				"type": "command"
			}, {
				"name": "Stop",
				"type": "command"
			}, {
				"name": "Accelerate",
				"type": "command"
			}]
		}
	},
	"browseServerOptions": null,
[... 234 lines cut ...]

  orion_mongo:
    hostname: orion_mongo
    image: mongo:4.2
    networks:
      - ocbnet
    ports:
      - "27017:27017"
    volumes:
      - orion_mongo_data:/data/db
      - orion_mongo_conf:/data/configdb
    command: --nojournal

volumes:
  iot_mongo_data:
  iot_mongo_conf:
  orion_mongo_data:
  orion_mongo_conf:

networks:
  hostnet:
  iotnet:
  ocbnet:

```

manfredipist, 2021-10-04:
Thanks for your help @manolait! From what i can see you're using NGSI-LD but trying to send requests using API for NGSIv2. If you don't really need ngsi-ld support, i'd suggest you to edit your docker-compose (comment  IOTA_CB_NGSI_VERSION environment and replace orion image name) and AGECONF/config,json files accordingly (replace your config,json with the content of the one called config_NGSIV2.json)

manolait, 2021-10-04:
Hello,

I have tried to make the changes and it gives error to raise the OCB server by the -forwarding command. I have removed it and the server is up but it still does not work {"error": "NotFound", "description": "The requested entity has not been found. Check type and id"}% 

Checking the database only creates Orion-opcua_car

Docker file
```

version: "3"

services:
  iotcarsrv:
    hostname: iotcarsrv
    image: iotagent4fiware/opcuacarsrv:latest
    volumes:
      - ./CARCONF/car_config.json:/opt/opc-ua-car-server/Car/Car1/config/car_config.json
    networks:
      - hostnet
    ports:
      - "5001:5001"

  iotage:
    hostname: iotage
    image: iotagent4fiware/iotagent-opcua:latest
    networks:
      - hostnet
      - iotnet
    ports:
      - "4001:4001"
      - "4081:8080"
    depends_on:
      - iotcarsrv
      - iotmongo
      - orion
    volumes:
      - ./AGECONF:/opt/iotagent-opcua/conf
      - ./certificates:/opt/iotagent-opcua/certificates
    environment:
      - IOTA_REGISTRY_TYPE=mongodb #Whether to hold IoT device info in memory or in a database
      - IOTA_LOG_LEVEL=DEBUG # The log level of the IoT Agent
      - IOTA_MONGO_HOST=iot_mongo # The host name of MongoDB
      - IOTA_MONGO_DB=iotagent_opcua # The name of the database used in mongoDB
      # - IOTA_CB_NGSI_VERSION=ld #comment if you mind using NGSIv2
      # - IOTA_JSON_LD_CONTEXT=https://uri.etsi.org/ngsi-ld/v1/ngsi-ld-core-context-v1.3.jsonld #comment if you mind using NGSIv2
      - IOTA_FALLBACK_TENANT=opcua_car
      - IOTA_RELAX_TEMPLATE_VALIDATION=true

  iotmongo:
    hostname: iot_mongo
    image: mongo:4.2
    networks:
      - iotnet
    volumes:
      - iot_mongo_data:/data/db
      - iot_mongo_conf:/data/configdb

  ################ OCB ################

  orion:
    hostname: orion
    image: fiware/orion:latest #replace with orion:latest if you mind using NGSIv2
    networks:
      - hostnet
      - ocbnet
    ports:
      - "1026:1026"
    depends_on:
      - orion_mongo
    command: -statCounters -dbhost orion_mongo -logLevel INFO

  orion_mongo:
    hostname: orion_mongo
    image: mongo:4.2
    networks:
      - ocbnet
    ports:
      - "27017:27017"
    volumes:
      - orion_mongo_data:/data/db
      - orion_mongo_conf:/data/configdb
    command: --nojournal

volumes:
  iot_mongo_data:
  iot_mongo_conf:
  orion_mongo_data:
  orion_mongo_conf:

networks:
  hostnet:
  iotnet:
  ocbnet:
```

config file
```

{
	"logLevel": "INFO",
	"multiCore": false,
	"relaxTemplateValidation": true,
	"contextBroker": {
		"host": "orion",
		"port": 1026,
		"service": "opcua_car",
		"subservice": "/demo"
	},
	"server": {
		"port": 4001,
		"baseRoot": "/"
	},
	"deviceRegistry": {
		"type": "memory"
	},
	"mongodb": {
		"host": "iot_mongo",
		"port": "27017",
		"db": "iotagent",
		"retries": 5,
		"retryTime": 5
	},
	"types": {
		"Device": {
			"service": "opcua_car",
			"subservice": "/demo",
			"active": [{
				"name": "Acceleration",
				"type": "Number"
			}, {
				"name": "EngineStopped",
				"type": "Boolean"
			}, {
				"name": "Engine_Temperature",
				"type": "Number"
			}, {
				"name": "Engine_Oxigen",
				"type": "Number"
			}],
			"lazy": [{
				"name": "Speed",
				"type": "Number"
			}],
			"commands": [{
				"name": "Error",
				"type": "command"
			}, {
				"name": "Stop",
				"type": "command"
			}, {
				"name": "Accelerate",
				"type": "command"
			}]
		}
	},
	"browseServerOptions": null,
	"service": "opcua_car",
	"subservice": "/demo",
	"providerUrl": "http://iotage:4001",
	"pollingExpiration": "200000",
	"pollingDaemonFrequency": "20000",
	"deviceRegistrationDuration": "P1M",
	"defaultType": null,
	"contexts": [{
		"id": "age01_Car",
		"type": "Device",
		"service": "opcua_car",
		"subservice": "/demo",
		"polling": false,
		"mappings": [{
			"ocb_id": "Error",
			"opcua_id": "ns=3;s=Error",
			"object_id": null,
			"inputArguments": []
		}, {
			"ocb_id": "Acceleration",
			"opcua_id": "ns=3;s=Acceleration",
			"object_id": null,
			"inputArguments": []
		}, {
			"ocb_id": "EngineStopped",
			"opcua_id": "ns=3;s=EngineStopped",
			"object_id": null,
			"inputArguments": []
		}, {
			"ocb_id": "Engine_Temperature",
			"opcua_id": "ns=3;s=Temperature",
			"object_id": null,
			"inputArguments": []
		}, {
			"ocb_id": "Engine_Oxigen",
			"opcua_id": "ns=3;s=Oxigen",
			"object_id": null,
			"inputArguments": []
		}]
	}],
	"contextSubscriptions": [{
		"id": "age01_Car",
		"type": "Device",
		"mappings": [{
			"ocb_id": "Error",
			"opcua_id": "ns=3;s=Error",
			"object_id": "ns=3;i=1000",
			"inputArguments": [{
				"dataType": 12,
				"type": "Error Type"
			}]
		}, {
			"ocb_id": "Speed",
			"opcua_id": "ns=3;s=Speed",
			"object_id": "ns=3;i=1000",
			"inputArguments": []
		}, {
			"ocb_id": "Stop",
			"opcua_id": "ns=3;s=Stop",
			"object_id": "ns=3;i=1000",
			"inputArguments": []
		}, {
			"ocb_id": "Accelerate",
			"opcua_id": "ns=3;s=Accelerate",
			"object_id": "ns=3;i=1000",
			"inputArguments": [{
				"dataType": 6,
				"type": "Intensity"
			}]
		}]
	}]
}
```

manfredipist, 2021-10-05:
Hello @manolait , i'm glad you sorted it out.
What do you mean for "the OPC server updates the data randomly or are they static?" ? Your can imagine the opcua server as a living object with its methods, attributes and initalizers. By using iotagent commands you are effectively sending a request to the server askingit to pursue an action. iotagent is subscribed to the the server and listen for changes actively or passively depending on your configuration.

manolait, 2021-10-06:
I have some questions

If for example we add a new device age02_Car in the IoTAgent. This does not appear in the OPC server since when the server was created only this age01_Car? We should create this new device in the server in the configuration file no?

When creating the server I don't understand the config file they use, what is the difference between contexts mappings and contextSubscriptions mappings?

Saludos, Manuel Muñoz Rodriguez