Engineering-Research-and-Development/iotagent-opcua #63: could not find the OCB context

manolait, 2021-11-04:
Hello,

I am having problems launching the IoTAgent opcua server. My config.json file theoretically works correctly since sometimes it works and sometimes it does not.

The log when the system is created is as follows:

if accessing the OPC as it retrieves the values of the nodes (ns=3;i=1006 value has changed to -1.334667) but when I query iot/devices/ it is empty but the OCB entities if it is filled.

```

iotage_1       | time=2021-11-04T13:05:38.174Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.Initialize | comp=iotAgent-OPCUA | srv=n/a | subsrv=n/a | msg= session created
iotage_1       | time=2021-11-04T13:05:38.176Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.Initialize | comp=iotAgent-OPCUA | srv=n/a | subsrv=n/a | msg= sessionId :  ns=1;g=2364B445-FE6E-4E42-9CA4-6A9B828545D2
iotage_1       | time=2021-11-04T13:05:38.176Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.Initialize | comp=iotAgent-OPCUA | srv=n/a | subsrv=n/a | msg= the timeout value set by the server is  100000  ms
iotage_1       | time=2021-11-04T13:05:38.525Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.Initialize | comp=iotAgent-OPCUA | srv=n/a | subsrv=n/a | msg= --- NAMESPACE ARRAY ---
iotage_1       | time=2021-11-04T13:05:38.526Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.Initialize | comp=iotAgent-OPCUA | srv=n/a | subsrv=n/a | msg= Namespace  0   :  http://opcfoundation.org/UA/
iotage_1       | time=2021-11-04T13:05:38.526Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.Initialize | comp=iotAgent-OPCUA | srv=n/a | subsrv=n/a | msg= Namespace  1   :  urn:MacBook-Pro.local:OPCUA:SimulationServer
iotage_1       | time=2021-11-04T13:05:38.526Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.Initialize | comp=iotAgent-OPCUA | srv=n/a | subsrv=n/a | msg= Namespace  2   :  http://www.prosysopc.com/OPCUA/SimulationServer/
iotage_1       | time=2021-11-04T13:05:38.526Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.Initialize | comp=iotAgent-OPCUA | srv=n/a | subsrv=n/a | msg= Namespace  3   :  http://www.prosysopc.com/OPCUA/SimulationNodes/
iotage_1       | time=2021-11-04T13:05:38.526Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.Initialize | comp=iotAgent-OPCUA | srv=n/a | subsrv=n/a | msg= Namespace  4   :  http://www.prosysopc.com/OPCUA/SimulationNodes/SimulationConfiguration/
iotage_1       | time=2021-11-04T13:05:38.526Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.Initialize | comp=iotAgent-OPCUA | srv=n/a | subsrv=n/a | msg= Namespace  5   :  null
iotage_1       | time=2021-11-04T13:05:38.526Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.Initialize | comp=iotAgent-OPCUA | srv=n/a | subsrv=n/a | msg= Namespace  6   :  null
iotage_1       | time=2021-11-04T13:05:38.527Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.Initialize | comp=iotAgent-OPCUA | srv=n/a | subsrv=n/a | msg= Namespace  7   :  urn:localhost/UA/CarServer
iotage_1       | time=2021-11-04T13:05:38.527Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.Initialize | comp=iotAgent-OPCUA | srv=n/a | subsrv=n/a | msg= Namespace  8   :  urn:localhost:UA:CarServer/SimulationConfiguration/
iotage_1       | time=2021-11-04T13:05:38.527Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.Initialize | comp=iotAgent-OPCUA | srv=n/a | subsrv=n/a | msg= Namespace  9   :  null
iotage_1       | time=2021-11-04T13:05:38.527Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.Initialize | comp=iotAgent-OPCUA | srv=n/a | subsrv=n/a | msg= Namespace  10   :  urn:localhost:5001/UA/CarServer/SimulationConfiguration/
iotage_1       | time=2021-11-04T13:05:38.527Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.Initialize | comp=iotAgent-OPCUA | srv=n/a | subsrv=n/a | msg= Namespace  11   :  urn:localhost/UA/CarServer/SimulationConfiguration/
iotage_1       | time=2021-11-04T13:05:38.527Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.Initialize | comp=iotAgent-OPCUA | srv=n/a | subsrv=n/a | msg= -----------------------
iotage_1       | time=2021-11-04T13:05:38.538Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.Initialize | comp=iotAgent-OPCUA | srv=n/a | subsrv=n/a | msg=registering OCB context React of type React
iotage_1       | time=2021-11-04T13:05:38.550Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.Initialize | comp=iotAgent-OPCUA | srv=sabana | subsrv=/ | msg=registering OCB context Simulation of type Simulation
iotage_1       | time=2021-11-04T13:05:38.558Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.Initialize | comp=iotAgent-OPCUA | srv=sabana | subsrv=/ | msg=Starting timer  -1
iotage_1       | time=2021-11-04T13:05:38.559Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.Initialize | comp=iotAgent-OPCUA | srv=sabana | subsrv=/ | msg=NO Timeout set!!!
iotage_1       | {"op":"IoTAgentNGSI.GenericMiddlewares","time":"2021-11-04T13:05:38.569Z","lvl":"DEBUG","msg":"Request for path [/iot/devices/React] from [localhost:4001]"}
iotage_1       | {"op":"IoTAgentNGSI.GenericMiddlewares","time":"2021-11-04T13:05:38.573Z","lvl":"DEBUG","msg":"Error [DEVICE_NOT_FOUND] handling request: No device was found with id:React"}
iotage_1       | {"op":"IoTAgentNGSI.DomainControl","time":"2021-11-04T13:05:38.581Z","lvl":"DEBUG","msg":"response-time: 14"}
iotage_1       | {"op":"IoTAgentNGSI.GenericMiddlewares","time":"2021-11-04T13:05:38.584Z","lvl":"DEBUG","msg":"Request for path [/iot/devices/Simulation] from [localhost:4001]"}
iotage_1       | {"op":"IoTAgentNGSI.GenericMiddlewares","time":"2021-11-04T13:05:38.585Z","lvl":"DEBUG","msg":"Error [DEVICE_NOT_FOUND] handling request: No device was found with id:Simulation"}
iotage_1       | {"op":"IoTAgentNGSI.DomainControl","time":"2021-11-04T13:05:38.588Z","lvl":"DEBUG","msg":"response-time: 4"}
iotage_1       | {"op":"IoTAgentNGSI.GenericMiddlewares","time":"2021-11-04T13:05:38.589Z","lvl":"DEBUG","msg":"Request for path [/iot/devices/React] from [localhost:4001]"}
iotage_1       | {"op":"IoTAgentNGSI.GenericMiddlewares","time":"2021-11-04T13:05:38.589Z","lvl":"DEBUG","msg":"Error [DEVICE_NOT_FOUND] handling request: No device was found with id:React"}
iotage_1       | {"op":"IoTAgentNGSI.DomainControl","time":"2021-11-04T13:05:38.590Z","lvl":"DEBUG","msg":"response-time: 1"}
iotage_1       | {"op":"IoTAgentNGSI.GenericMiddlewares","time":"2021-11-04T13:05:38.591Z","lvl":"DEBUG","msg":"Request for path [/iot/devices/Simulation] from [localhost:4001]"}
iotage_1       | {"op":"IoTAgentNGSI.GenericMiddlewares","time":"2021-11-04T13:05:38.591Z","lvl":"DEBUG","msg":"Error [DEVICE_NOT_FOUND] handling request: No device was found with id:Simulation"}
iotage_1       | {"op":"IoTAgentNGSI.DomainControl","time":"2021-11-04T13:05:38.592Z","lvl":"DEBUG","msg":"response-time: 1"}
iotage_1       | time=2021-11-04T13:05:38.597Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.Initialize | comp=iotAgent-OPCUA | srv=sabana | subsrv=/ | msg=device deleted!
iotage_1       | {"op":"IoTAgentNGSI.InMemoryGroupRegister","time":"2021-11-04T13:05:38.598Z","lvl":"DEBUG","msg":"Looking for device params [\"service\",\"subservice\",\"type\"]"}
iotage_1       | {"op":"IoTAgentNGSI.DeviceService","time":"2021-11-04T13:05:38.599Z","lvl":"DEBUG","msg":"deviceData after merge with conf: {\"id\":\"React\",\"name\":\"React\",\"type\":\"React\",\"active\":[{\"name\":\"Temperature\",\"type\":\"Number\"},{\"name\":\"Co2\",\"type\":\"Number\"},{\"name\":\"Radiation\",\"type\":\"Number\"}],\"service\":\"sabana\",\"subservice\":\"/\",\"polling\":false,\"endpoint\":\"opc.tcp://MacBook-Pro.local:5001/UA/CarServer\"}"}
iotage_1       | {"op":"IoTAgentNGSI.DeviceService","time":"2021-11-04T13:05:38.600Z","lvl":"DEBUG","msg":"deviceData before merge with conf: {\"id\":\"React\",\"name\":\"React\",\"type\":\"React\",\"active\":[{\"name\":\"Temperature\",\"type\":\"Number\",\"object_id\":\"Temperature\"},{\"name\":\"Co2\",\"type\":\"Number\",\"object_id\":\"Co2\"},{\"name\":\"Radiation\",\"type\":\"Number\",\"object_id\":\"Radiation\"}],\"lazy\":[],\"commands\":[],\"service\":\"sabana\",\"subservice\":\"/\",\"polling\":false,\"endpoint\":\"opc.tcp://MacBook-Pro.local:5001/UA/CarServer\",\"staticAttributes\":[],\"subscriptions\":[]}"}
iotage_1       | {"op":"IoTAgentNGSI.DeviceService","time":"2021-11-04T13:05:38.600Z","lvl":"DEBUG","msg":"Registering device into NGSI Service:\n{\n    \"id\": \"React\",\n    \"name\": \"React\",\n    \"type\": \"React\",\n    \"active\": [\n        {\n            \"name\": \"Temperature\",\n            \"type\": \"Number\",\n            \"object_id\": \"Temperature\"\n        },\n        {\n            \"name\": \"Co2\",\n      [... cut ...]
iotage_1       | time=2021-11-04T13:06:08.583Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.Monitoring | comp=iotAgent-OPCUA | srv=sabana | subsrv=/ | msg=ns=3;i=1010  value has changed to 27
```

```
time=2021-11-04T12:55:41.565Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.Monitoring | comp=iotAgent-OPCUA | srv=sabana | subsrv=/ | msg=ns=3;i=1006 value has changed to -1.334667

time=2021-11-04T12:55:41.565Z | lvl=ERROR | corr=n/a | trans=n/a | op=Index.Monitoring | comp=iotAgent-OPCUA | srv=sabana | subsrv=/ | msg=could not find the OCB context Simulation
```

MattiaMarzano-Eng, 2021-11-09:
Please can you provide the config.json and config.properties files you're using?

manolait, 2021-11-09:
I solved the problem.

It was related to the type in the context and mapping.

closed 2021-11-09 by manolait