Engineering-Research-and-Development/iotagent-opcua #64: Use of commands status PENDING

manolait, 2021-11-09:
hello,

I am trying to use the commands to send information to my OPC server. I have a number of doubts.

Doubt 1: When I send my update (why is it necessary to put it between [])

curl -X PUT \
  'http://localhost:1026/v2/entities/React/attrs/Imput?type=3:React' \
  -H 'content-type: application/json' \
  -H 'fiware-service: sabana' \
  -H 'fiware-servicepath: /team' \
  -d '{
  "value": [44],
  "type": "command"
}’

Doubt 2: When I query my OCB I get the response from that entity and the command attribute to update is PENDING. How do I solve this problem.

```
"Imput_info": {
            "type": "commandResult",
            "value": " ",
            "metadata": {}
        },
        "Imput_status": {
            "type": "commandStatus",
            "value": "PENDING",
            "metadata": {}
        }
```

Thank you.

MattiaMarzano-Eng, 2021-11-09:
Are you using the same set up of the [issue #63](https://github.com/Engineering-Research-and-Development/iotagent-opcua/issues/63)?

manolait, 2021-11-09:
Hello,

this is my system configuration

Config.properties
```

## SOUTHBOUND CONFIGURATION (OPC UA)
namespace-ignore=2,7
endpoint=opc.tcp://MacBook-Pro.local:5001/UA/CarServer

## NORTHBOUND CONFIGURATION (ORION CONTEXT BROKER)
context-broker-host=orion
context-broker-port=1026
fiware-service=sabana
fiware-service-path=/team

## AGENT CONFIGURATION
server-base-root=/
server-port=4001
provider-url=http://iotage:4001

device-registration-duration=P1M
device-registry-type=memory

log-level=DEBUG

namespaceIndex=3
namespaceNumericIdentifier=1000

# MONGO-DB CONFIGURATION (required if device-registry-type=mongodb)
mongodb-host=iotmongo
mongodb-port=27017
mongodb-db=iotagent
mongodb-retries=5
mongodb-retry-time=5

## DATATYPE MAPPING OPCUA --> NGSI
OPC-datatype-Number=Number
OPC-datatype-Decimal128=Number
OPC-datatype-Double=Number
OPC-datatype-Float=Number
OPC-datatype-Integer=Integer
OPC-datatype-UInteger=Integer
OPC-datatype-String=Text
OPC-datatype-ByteString=Text
#END DATATYPE MAPPING OPCUA --> NGSI

## SESSION PARAMETERS
requestedPublishingInterval=10
requestedLifetimeCount=1000
requestedMaxKeepAliveCount=10
maxNotificationsPerPublish=100
publishingEnabled=true
priority=10

#SubscriptionsStrategy
uniqueSubscription=true

## MONITORING PARAMETERS
samplingInterval=1
queueSize=10000
discardOldest=false

## SERVER CERT E AUTH
securityMode=None
securityPolicy=None
userName=
password=

#securityMode=SIGNANDENCRYPT
#securityPolicy=1Basic256
#password=password1
#userName=user1

#api-ip=192.168.13.153

## ADMINISTRATION SERVICES
api-port=8080

## POLL COMMANDS SETTINGS
polling=false
polling-commands-timer=30000
pollingDaemonFrequency=20000
pollingExpiration=200000

## AGENT ID
# agent-id=age01_
# entity-id=age01_Car # used only during tests

## CONFIGURATION
# configuration=api

## CHECK TIMER POLLING DEVICES
checkTimer=2000

```

config.json
```

{
	"logLevel": "INFO",
	"multiCore": false,
	"relaxTemplateValidation": true,
	"contextBroker": {
		"host": "orion",
		"port": 1026,
		"service": "opcua_car",
		"subservice": "/demo"
	},
	"server": {
		"port": 4001,
		"baseRoot": "/"
	},
	"deviceRegistry": {
		"type": "memory"
	},
	"mongodb": {
		"host": "iot_mongo",
		"port": "27017",
		"db": "iotagent",
		"retries": 5,
		"retryTime": 5
	},
	"types": {
		"Device": {
			"service": "opcua_car",
			"subservice": "/demo",
			"active": [{
				"name": "Acceleration",
				"type": "Number"
			}, {
				"name": "EngineStopped",
				"type": "Boolean"
			}, {
				"name": "Engine_Temperature",
				"type": "Number"
			}, {
				"name": "Engine_Oxigen",
				"type": "Number"
			}],
			"lazy": [{
				"name": "Speed",
				"type": "Number"
			}],
			"commands": [{
				"name": "Error",
				"type": "command"
			}, {
				"name": "Stop",
				"type": "command"
			}, {
				"name": "Accelerate",
				"type": "command"
			}]
		}
	},
	"browseServerOptions": null,
	"service": "opcua_car",
	"subservice": "/demo",
	"providerUrl": "http://iotage:4001",
	"pollingExpiration": "200000",
	"pollingDaemonFrequency": "20000",
	"deviceRegistrationDuration": "P1M",
	"defaultType": null,
	"contexts": [{
		"id": "age01_Car",
		"type": "Device",
		"service": "opcua_car",
		"subservice": "/demo",
		"polling": false,
		"mappings": [{
			"ocb_id": "Error",
			"opcua_id": "ns=3;s=Error",
			"object_id": null,
			"inputArguments": []
		}, {
			"ocb_id": "Acceleration",
			"opcua_id": "ns=3;s=Acceleration",
			"object_id": null,
			"inputArguments": []
		}, {
			"ocb_id": "EngineStopped",
			"opcua_id": "ns=3;s=EngineStopped",
			"object_id": null,
			"inputArguments": []
		}, {
			"ocb_id": "Engine_Temperature",
			"opcua_id": "ns=3;s=Temperature",
			"object_id": null,
			"inputArguments": []
		}, {
			"ocb_id": "Engine_Oxigen",
			"opcua_id": "ns=3;s=Oxigen",
			"object_id": null,
			"inputArguments": []
		}]
	}],
	"contextSubscriptions": [{
		"id": "age01_Car",
		"type": "Device",
		"mappings": [{
			"ocb_id": "Error",
			"opcua_id": "ns=3;s=Error",
			"object_id": "ns=3;i=1000",
			"inputArguments": [{
				"dataType": 12,
				"type": "Error Type"
			}]
		}, {
			"ocb_id": "Speed",
			"opcua_id": "ns=3;s=Speed",
			"object_id": "ns=3;i=1000",
			"inputArguments": []
		}, {
			"ocb_id": "Stop",
			"opcua_id": "ns=3;s=Stop",
			"object_id": "ns=3;i=1000",
			"inputArguments": []
		}, {
			"ocb_id": "Accelerate",
			"opcua_id": "ns=3;s=Accelerate",
			"object_id": "ns=3;i=1000",
			"inputArguments": [{
				"dataType": 6,
				"type": "Intensity"
			}]
		}]
	}]
}
```

MattiaMarzano-Eng, 2021-11-10:
It could be, please share also your server setting and object folder

manolait, 2021-11-15:
Hello,

I am modifying all my configuration attached the information to be OPC server, config.jon, config.properties

Opc where my command would be the variable2
```

const { OPCUAServer, Variant, DataType, StatusCodes } = require("node-opcua");

(async () => {

    // Let's create an instance of OPCUAServer
    const server = new OPCUAServer({
        port: 5001, // the port of the listening socket of the server
        resourcePath: "/UA/CarServer", // this path will be added to the endpoint resource name
        buildInfo: {
            productName: "CarServer",
            buildNumber: "7658",
            buildDate: new Date(2014, 5, 2)
        }
    });
    await server.initialize();
    console.log("initialized");

    const addressSpace = server.engine.addressSpace;
    const namespace = addressSpace.getOwnNamespace();

    // declare a new object
    const device = namespace.addObject({
        organizedBy: addressSpace.rootFolder.objects,
        browseName: "MyDevice"
    });

    // add some variables
    // add a variable named MyVariable1 to the newly created folder "MyDevice"
    let variable1 = 1;

    // emulate variable1 changing every 500 ms
    setInterval(() => { variable1 += 1; }, 500);

    namespace.addVariable({
        componentOf: device,
        nodeId: "ns=1;s=Variable1", // some opaque NodeId in namespace 4
        browseName: "Variable1",
        dataType: "Double",
        value: {
            get: () => new Variant({ dataType: DataType.Double, value: variable1 })
        }
    });

    // add a variable named MyVariable2 to the newly created folder "MyDevice"
    let variable2 = 10.0;

    namespace.addVariable({

        componentOf: device,

        nodeId: "ns=1;s=Variable2", // some opaque NodeId in namespace 4

        browseName: "Variable2",

        dataType: "Double",

        value: {
            get: () => new Variant({ dataType: DataType.Double, value: variable2 }),
            set: (variant) => {
                variable2 = parseFloat(variant.value);
                return StatusCodes.Good;
            }
        }
[... 238 lines cut ...]
Fiware entities
```

[
    {
        "id": "MyDevice",
        "type": "MyDevice",
        "Variable1": {
            "type": "Number",
            "value": 254,
            "metadata": {}
        },
        "Variable2_info": {
            "type": "commandResult",
            "value": " ",
            "metadata": {}
        },
        "Variable2_status": {
            "type": "commandStatus",
            "value": "UNKNOWN",
            "metadata": {}
        }
    }
]
```

manolait, 2021-11-23:
Hello,

could you help me with the problem? 

Best regards.

MattiaMarzano-Eng, 2022-05-27:
Please try to use the latest version of the docker image now. Leave a comment if the issue persist.

manfredipist, 2023-01-12:
Hi @manolait , command execution flow has been totally revised on v2 of the agent thus i suggest you to try the latest version to check if the issue still persists.

closed 2023-01-12 by manfredipist