Engineering-Research-and-Development/iotagent-opcua #65: Not update the conf.properties once modified of the docker

IsabelManiega, 2021-11-22:
Hi! 
I try it to use the lastest version iotagent_opcua in Fiware, my docker-compose.yml is this:

version: "3.5"
secrets:
    age_idm_auth:
        file: age_idm_auth.txt

services:
  # Orion is the context broker
  orion:
    image: fiware/orion:latest
    hostname: orion
    container_name: fiware-orion
    depends_on:
      - mongo-db
    networks:
      - fiware_default
      - hostnet
    expose:
      - "1026"
    ports:
      - "1026:1026" # localhost:1026
    command:  -statCounters -dbhost mongo-db -logForHumans -logLevel DEBUG -t 255
    healthcheck:
      test: curl --fail -s http://orion:1026/version || exit 1
      interval: 5s

  iot-agent:
    image: fiware/iotagent-ul:latest
    hostname: iot-agent
    container_name: fiware-iot-agent
    depends_on:
      - mongo-db
      - mosquitto
    networks:
      - fiware_default
    expose:
      - "4041"
      - "7896"
    ports:
      - "4041:4041" # localhost:4041
      - "7896:7896"
    environment:
      - IOTA_CB_HOST=orion # name of the context broker to update context
      - IOTA_CB_PORT=1026 # port the context broker listens on to update context
      - IOTA_NORTH_PORT=4041
      - IOTA_REGISTRY_TYPE=mongodb #Whether to hold IoT device info in memory or in a database
      - IOTA_LOG_LEVEL=DEBUG # The log level of the IoT Agent
      - IOTA_TIMESTAMP=true # Supply timestamp information with each measurement
      - IOTA_CB_NGSI_VERSION=v2 # use NGSIv2 when sending updates for active attributes
      - IOTA_AUTOCAST=true # Ensure Ultralight number values are read as numbers not strings
      - IOTA_MONGO_HOST=mongo-db # The host name of MongoDB
      - IOTA_MONGO_PORT=27017 # The port mongoDB is listening on
      - IOTA_MONGO_DB=iotagentul # The name of the database used in mongoDB
      - IOTA_HTTP_PORT=7896
      - IOTA_MQTT_HOST=mosquitto # The host name of the MQTT Broker
      - IOTA_MQTT_PORT=1883 # The port the MQTT Broker is listening on to receive topics
      # - IOTA_DEFAULT_RESOURCE= # Default is blank. I'm using MQTT so I don't need a resource
      - IOTA_PROVIDER_URL=http://iot-agent:4041
    healthcheck:
      interval: 5s

  # Tutorial acts as a series of dummy IoT Sensors over HTTP
  tutorial:
    image: fiware/tutorials.context-provider
    hostname: tutorial
    container_name: fiware-tutorial
    depends_on:
      - orion
[... 387 lines cut ...]
           "inputArguments":[

           ]
        },
        {
           "ocb_id":"current",
           "opcua_id":"ns=2;s=current_scope_rms",
           "object_id":null,
           "inputArguments":[

           ]
        },
        {
           "ocb_id":"stop_process",
           "opcua_id":"ns=2;s=stop_process",
           "object_id":null,
           "inputArguments":[

           ]
        }]
	}]
}
The problem is when I run the docker-compose, the iotagent shows always the default conf.properties, it try to connect to opc.tcp://iotcarsrv:5001/UA/CarServer not my config.properties.

Thank you in advance,

MattiaMarzano-Eng, 2021-11-22:
Hello @IsabelManiega, please try to use `iotagent4fiware/iotagent-opcua:1.3.4` in your docker-compose

IsabelManiega, 2021-11-22:
Thank you very much for your prompt response!
Yes, It is working with version 1.3.4.

closed 2021-11-22 by MattiaMarzano-Eng