Engineering-Research-and-Development/iotagent-opcua #69: Problem with connecting to OPC server

piotr-paul, 2022-03-08:
Hi,

I'm trying to connect the agent to the Bystronic OPC server.
Agent connected well (config was generated), but I have problems with registering a new device into it. How should I do this?

MattiaMarzano-Eng, 2022-03-08:
Please double check the issue #68 or [this part of documentation](https://iotagent-opcua.readthedocs.io/en/latest/opc_ua_agent_tutorial/index.html#step-4-provision-a-new-device)

drypatrick, 2022-05-12:
@MattiaMarzano-Eng I'm struggling with a similar issue, trying to add a device.
If I run the local OPCUA server, running the docker-compse.yml, I'm able to load a device with the REST API
`curl http://localhost:4001/iot/devices -H "fiware-service: opcua_car" -H "fiware-servicepath: /demo" -H "Content-Type: application/json" -d @testCommands/add_device_NGSIv2.json`
Then, if I run the docker-compse-external-server.yml, which uses the iotagent4fiware/iotagent-opcua:1.3.4, I get the following error `{"name":"WRONG_SYNTAX","message":"Wrong syntax in request: Errors found validating request."}`

From the Docker log:

```
"op":"IoTAgentNGSI.RestUtils","time":"2022-05-12T19:47:59.467Z","lvl":"DEBUG","msg":"Errors found validating request: {
	"valid":false,"errors": [
		{"attribute":"pattern","property":"devices.0.commands.1.object_id","expected":"^([^<>();'="]+)+$","actual":"ns=3;s=Accelerate","message":"invalid input"},
		{"attribute":"pattern","property":"devices.0.commands.0.object_id","expected":"^([^<>();'="]+)+$","actual":"ns=3;s=Stop","message":"invalid input"},
		{"attribute":"pattern","property":"devices.0.attributes.4.object_id","expected":"^([^<>();'="]+)+$","actual":"ns=3;s=Oxigen","message":"invalid input"},
		{"attribute":"pattern","property":"devices.0.attributes.3.object_id","expected":"^([^<>();'="]+)+$","actual":"ns=3;s=Temperature","message":"invalid input"},
		{"attribute":"pattern","property":"devices.0.attributes.2.object_id","expected":"^([^<>();'="]+)+$","actual":"ns=3;s=EngineStopped","message":"invalid input"},
		{"attribute":"pattern","property":"devices.0.attributes.1.object_id","expected":"^([^<>();'="]+)+$","actual":"ns=3;s=Acceleration","message":"invalid input"},
		{"attribute":"pattern","property":"devices.0.attributes.0.object_id","expected":"^([^<>();'="]+)+$","actual":"ns=3;s=EngineBrake","message":"invalid input"},
		{"attribute":"pattern","property":"devices.0.lazy.0.object_id","expected":"^([^<>();'="]+)+$","actual":"ns=3;s=Speed","message":"invalid input"}
	]
}
```

MattiaMarzano-Eng, 2022-05-26:
Docker image updated, please doublechek if the issue persist.

drypatrick, 2022-05-26:
With the new image the syntax problem is solved, but I am encountering another problem always in adding a device via API.
Most often running the command:
```
curl -iX POST \
> 'http: // localhost: 4001 / iot / devices' \
> -H 'fiware-service: opcua_car' \
> -H 'fiware-servicepath: / demo' \
> -H 'Content-Type: application / json' \
> -d @ CONFIGURATOR / add_device_config.json
```
I get this answer from the server:
```
curl: (52) Empty reply from server`
```
But without any change in the `CONFIGURATOR/add_device_config.json` sometimes it works, and I get the answer:
```
HTTP / 1.1 201 Created
X-Powered-By: Express
Fiware-Correlator: 596ec14f-49bd-464a-bd12-71cfba786997
Content-Type: application / json; charset = utf-8
Content-Length: 2
ETag: W / "2-vyGp6PvFo4RvsFtPoIWeCReyIC8"
Date: Thu, 26 May 2022 19:07:24 GMT
Connection: keep-alive
Keep-Alive: timeout = 5
```
I can't understand why I go through all these difficulties using the REST API, manually populating the `config.json` seems to be working so far.

vonbrand, 2022-12-19:
> Please double check the issue #68 or [this part of documentation](https://iotagent-opcua.readthedocs.io/en/latest/opc_ua_agent_tutorial/index.html#step-4-provision-a-new-device)

Nothing listening on 4001 here.

closed 2023-01-12 by manfredipist