Engineering-Research-and-Development/iotagent-opcua #71: External OPC UA server cert auth

petermatyas, 2022-04-21:
Hello,

I try to connect an external OPC UA server. Unefortunetly the certificate of the server is expired. When I open it a 3rd party software I can accept it, and I can see the data. 
I'd like to use the mapping tool for the configuration of the iotagent. When I execute it, the mapping tool return with error. It execute the `java -jar mapping_tool.jar  -e opc.tcp://130.130.130.1:4840 -f conf/config.properties -u null -p null` code

The log:
```
iotage_1       | ........time=2022-04-21T07:58:49.685Z | lvl=ERROR | corr=n/a | trans=n/a | op=Index.MappingTool | comp=iotAgent-OPCUA | srv=n/a | subsrv=n/a | msg=There is a problem with automatic configuration. Loading old configuration (if exists)...Error: Command failed: java -jar mapping_tool.jar  -e opc.tcp://130.130.130.1:4840 
-f conf/config.properties -u null -p null
iotage_1       | [main] INFO org.opcfoundation.ua.utils.CryptoUtil - SecurityProvider initialized from org.bouncycastle.jce.provider.BouncyCastleProvider
iotage_1       | [main] INFO org.opcfoundation.ua.utils.CryptoUtil - Using SecurityProvider BC
iotage_1       | [main] INFO org.opcfoundation.ua.transport.tcp.io.TcpConnection - /130.130.130.1:4840 Connecting
iotage_1       | [main] INFO org.opcfoundation.ua.transport.tcp.io.TcpConnection - /130.130.130.1:4840 Connected
iotage_1       | [main] INFO org.opcfoundation.ua.transport.tcp.io.SecureChannelTcp - 1735193229 Closed
iotage_1       | [main] INFO org.opcfoundation.ua.transport.tcp.io.TcpConnection - /130.130.130.1:4840 Closed
iotage_1       | [TcpConnection/Read] INFO org.opcfoundation.ua.transport.tcp.io.TcpConnection - /130.130.130.1:4840 Closed (expected)
iotage_1       | [main] INFO org.opcfoundation.ua.transport.tcp.io.TcpConnection - sif401:4840 Connecting
iotage_1       | [main] INFO org.opcfoundation.ua.transport.tcp.io.TcpConnection - sif401:4840 Connect failed
iotage_1       | java.net.UnknownHostException: sif401
iotage_1       |        at java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:184)
iotage_1       |        at java.net.SocksSocketImpl.connect(SocksSocketImpl.java:392)
iotage_1       |        at java.net.Socket.connect(Socket.java:607)
iotage_1       |        at org.opcfoundation.ua.transport.tcp.io.TcpConnection.open(TcpConnection.java:431)
iotage_1       |        at org.opcfoundation.ua.transport.tcp.io.SecureChannelTcp.open(SecureChannelTcp.java:566)
iotage_1       |        at org.opcfoundation.ua.application.Client.createSecureChannel(Client.java:644)
iotage_1       |        at org.opcfoundation.ua.application.Client.createSecureChannel(Client.java:558)
iotage_1       |        at org.opcfoundation.ua.application.Client.createSecureChannel(Client.java:541)
iotage_1       |        at MappingTool.main(MappingTool.java:183)
iotage_1       | [main] WARN org.opcfoundation.ua.transport.tcp.io.SecureChannelTcp - Connection failed: Bad_ConnectionRejected (code=0x80AC0000, description="2158755840, sif401")
iotage_1       | org.opcfoundation.ua.common.ServiceResultException: Bad_ConnectionRejected (code=0x80AC0000, description="2158755840, sif401")
iotage_1       |        at org.opcfoundation.ua.transport.tcp.io.TcpConnection.open(TcpConnection.java:438)
iotage_1       |        at org.opcfoundation.ua.transport.tcp.io.SecureChannelTcp.open(SecureChannelTcp.java:566)
iotage_1       |        at org.opcfoundation.ua.application.Client.createSecureChannel(Client.java:644)
iotage_1       |        at org.opcfoundation.ua.application.Client.createSecureChannel(Client.java:558)
iotage_1       |        at org.opcfoundation.ua.application.Client.createSecureChannel(Client.java:541)
iotage_1       |        at MappingTool.main(MappingTool.java:183)
iotage_1       | Caused by: java.net.UnknownHostException: sif401
iotage_1       |        at java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:184)
iotage_1       |        at java.net.SocksSocketImpl.connect(SocksSocketImpl.java:392)
iotage_1       |        at java.net.Socket.connect(Socket.java:607)
iotage_1       |        at org.opcfoundation.ua.transport.tcp.io.TcpConnection.open(TcpConnection.java:431)
iotage_1       |        ... 5 more
iotage_1       |
iotage_1       | time=2022-04-21T07:58:51.039Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.Initialize | comp=iotAgent-OPCUA | srv=n/a | subsrv=n/a | msg=----------------------------------------------------
iotage_1       | time=2022-04-21T07:58:51.040Z | lvl=INFO | corr=n/a | trans=n/a | op=Index.Initialize | comp=iotAgent-OPCUA | srv=n/a | subsrv=n/a | msg=endpointUrl   
      =  opc.tcp://130.130.130.1:4840
iotage_1       | {"op":"IoTAgentNGSI.DomainControl","time":"2022-04-21T07:58:51.045Z","lvl":"ERROR","msg":"{ TypeError: Cannot read property 'toString' of null\n    at 
Object.run (/opt/iotagent-opcua/iot_agent_modules/run/run.js:92:72)\n    at /opt/iotagent-opcua/index.js:90:21\n    at ChildProcess.exithandler (child_process.js:301:5)\n    at ChildProcess.emit (events.js:198:13)\n    at maybeClose (internal/child_process.js:982:16)\n    at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)\n  domain:\n   Domain {\n     domain: null,\n     _events:\n      [Object: null prototype] {\n        removeListener: [Function: updateExceptionCapture],\n   
     newListener: [Function: updateExceptionCapture],\n        error: [Function: domainErrorHandler] },\n     _eventsCount: 3,\n     _maxListeners: undefined,\n     members: [],\n     start: 1650527930740,\n     trans: '372b0a80-5a27-4cfa-a418-3ba4965c1643',\n     corr: '372b0a80-5a27-4cfa-a418-3ba4965c1643',\n     op: 'IoTAgentNGSI.DeviceService',\n     [Symbol(kWeak)]: WeakReference {} },\n  domainThrown: true }"}
iotagent-opcua_iotage_1 exited with code 
```

When I execute the mapping tool manually without user and password, I get one exception, but the mapping run succesfully.
`java -jar mapping_tool.jar -e opc.tcp://130.130.130.1:4840 -f conf\config.properties -u  -p  `

```
java -jar mapping_tool.jar -e opc.tcp://130.130.130.1:4840 -f conf\config.properties -u  -p 
[INFO ] 2022-04-21 10:02:43.672 [MappingTool:88] - Welcome to ENGINEERING INGEGNERIA INFORMATICA FIWARE OPC UA AGENT MAPPING TOOL
[INFO ] 2022-04-21 10:02:47.362 [TcpConnection:488] - /130.130.130.1:4840 Connecting
[INFO ] 2022-04-21 10:02:47.425 [TcpConnection:712] - Connected (non-reverse), handshake completed, local=/130.130.130.205:19119, remote=/130.130.130.1:4840
[INFO ] 2022-04-21 10:02:47.523 [SecureChannelTcp:870] - 1735193230 Closed
[INFO ] 2022-04-21 10:02:47.529 [TcpConnection:823] - /130.130.130.1:4840 Closed
[INFO ] 2022-04-21 10:02:47.529 [TcpConnection:1171] - /130.130.130.1:4840 Closed (expected)
[INFO ] 2022-04-21 10:02:47.535 [TcpConnection:488] - /130.130.130.1:4840 Connecting        
[INFO ] 2022-04-21 10:02:47.586 [TcpConnection:712] - Connected (non-reverse), handshake completed, local=/130.130.130.205:19120, remote=/130.130.130.1:4840
java.lang.ArrayIndexOutOfBoundsException: 1
        at AddressSpaceBrowsing.browse(AddressSpaceBrowsing.java:145)
        at MappingTool.main(MappingTool.java:410)
[INFO ] 2022-04-21 10:03:12.211 [AddressSpaceBrowsing:338] - ObjectType: AggregateConfigurationType(ns=0;i=11187) --> i=0
|____[INFO ] 2022-04-21 10:03:12.233 [AddressSpaceBrowsing:338] - Variable: PercentDataBad(ns=0;i=11189) --> i=68
|____|____[INFO ] 2022-04-21 10:03:12.253 [AddressSpaceBrowsing:338] - Object: Mandatory(ns=0;i=78) --> i=77
|____|____|____[INFO ] 2022-04-21 10:03:12.279 [AddressSpaceBrowsing:338] - ObjectType: ModellingRuleType(ns=0;i=77) --> i=0
|____|____|____|____[INFO ] 2022-04-21 10:03:12.302 [AddressSpaceBrowsing:338] - Variable: NamingRule(ns=0;i=111) --> i=68
|____|____|____|____|____[INFO ] 2022-04-21 10:03:12.325 [AddressSpaceBrowsing:338] - Object: Mandatory(ns=0;i=78) --> i=77
|____|____|____[INFO ] 2022-04-21 10:03:12.334 [AddressSpaceBrowsing:338] - Variable: NamingRule(ns=0;i=112) --> i=68
|____[INFO ] 2022-04-21 10:03:12.352 [AddressSpaceBrowsing:338] - Variable: PercentDataGood(ns=0;i=11190) --> i=68
```

In the config the security mode and policy is None, and I don't fill the user and the password.
```
## SERVER CERT E AUTH
securityMode=None
securityPolicy=None
userName= 
password= 
```

I'm nut sure if it is a solution or not, but if the security mode is None, could be the mapping tool parameter an empty caharcer, not null?

Thank you!
Péter

MattiaMarzano-Eng, 2022-04-21:
Hello Péter,
thank for your feedback. We're working at a new version of the mapping tool ( you can find it in the dedicated branch) that is under development, so not yet available. Btw my hint is to proceed like you did, run the mapping tool manually, get the config file and use it in the agent. We will take care of this issue in the development of the new mapping tool will be released soon.

closed 2022-04-21 by MattiaMarzano-Eng

petermatyas, 2022-04-21:
Hello,

Thank you for your support!

Péter