Engineering-Research-and-Development/iotagent-opcua #72: debuging

petermatyas, 2022-04-26:
Hello,

Sorry for my stupid question, I'm not familiar with NodJS, but I would like to find what cause my error message. 
My problem is, I can't see my error messages on the console after I execute the docker file.

I tried these in the run.js file, even if I wrtite after the `logger.info(logContext, '----------------------------------------------------');` line:
`logger.info(logContext, '--error--');`
`console.log('--error--');`

 In the config.properties the `log-level=INFO`.

I use this command for docker:
`docker-compose -f docker-compose-external-server.yml up --build`
I tried the `--force-recreate flag`, but nothing changed.

Thank you,
Péter

MattiaMarzano-Eng, 2022-05-20:
Hello Péter,
I'm sorry but it's not totally clear to me your question.
Are you running the agent using docker? in that case you can't edit the run.js file, or better you can do it but your changes don't affect the running agent.

petermatyas, 2022-05-26:
Hello MattiaMarzano-Eng,
Hello drypatrick

Thank you! Drypatrick has right, I had to execute only the nodeJS. My original error message was from docker: 
```
time=2022-05-26T09:38:05.454Z | lvl=ERROR | corr=n/a | trans=n/a | op=n/a | msg=TypeError: Cannot read property 'toString' of null
```
what say nothing. Execuding directly I found where stop the program:
```
time=2022-05-26T10:50:36.896Z | lvl=FATAL | corr=n/a | trans=n/a | op=IoTAgentNGSI.Global | from=n/a | srv=n/a | subsrv=n/a | msg=An unexpected exception has been raised. Ignoring: TypeError: Cannot read property '0' of undefined
    at C:\Users\peter\Documents\projektek\fiware_opcua_test\iotagent-opcua\iot_agent_modules\run\run.js:1388:42
    at Array.forEach (<anonymous>)
    at Timeout._onTimeout (C:\Users\peter\Documents\projektek\fiware_opcua_test\iotagent-opcua\iot_agent_modules\run\run.js:1385:38)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7) | comp=IoTAgent
time=2022-05-26T10:50:36.899Z | lvl=ERROR | corr=n/a | trans=n/a | op=n/a | msg=TypeError: Cannot read property '0' of undefined
```

This is an existing OPC-UA server. I can read the OPC-UA data with 3rd party software without error. The config.json is created by the mapping_tool.jar. 

Péter

drypatrick, 2022-05-26:
Also I have problems with the mapping tool, especially the tool is able to map namespaces 1, 2, 3, but not the 4 where all my variables reside.

Here's my `config.properties`:
```
## SOUTHBOUND CONFIGURATION (OPC UA)
namespace-ignore=0,1,2,3
endpoint=opc.tcp://iotcarsrv:4840/

# ################ OPTIONAL FILTERING ################
# nodes-filtering-in=ns\=3;i\=6010,ns\=1;s\=Speed
# nodes-filtering-out=ns\=1;i\=1,ns\=1;s\=Temperature
# ####################################################

## NORTHBOUND CONFIGURATION (ORION CONTEXT BROKER)
context-broker-host=orion
context-broker-port=1026
fiware-service=opcua_car
fiware-service-path=/demo

## AGENT CONFIGURATION
server-base-root=/
server-port=4001
provider-url=http://iotage:4001

device-registration-duration=P1M
device-registry-type=memory

log-level=DEBUG

namespaceIndex=4
namespaceNumericIdentifier=1

# MONGO-DB CONFIGURATION (required if device-registry-type=mongodb)
mongodb-host=iotmongo
mongodb-port=27017
mongodb-db=iotagent
mongodb-retries=5
mongodb-retry-time=5

## DATATYPE MAPPING OPCUA --> NGSI
OPC-datatype-Number=Number
OPC-datatype-Decimal128=Number
OPC-datatype-Double=Number
OPC-datatype-Float=Number
OPC-datatype-Integer=Integer
OPC-datatype-UInteger=Integer
OPC-datatype-String=Text
OPC-datatype-ByteString=Text
#END DATATYPE MAPPING OPCUA --> NGSI

## SESSION PARAMETERS
requestedPublishingInterval=10
requestedLifetimeCount=1000
requestedMaxKeepAliveCount=10
maxNotificationsPerPublish=100
publishingEnabled=true
priority=10

#SubscriptionsStrategy
uniqueSubscription=false

## MONITORING PARAMETERS
samplingInterval=1
queueSize=10000
discardOldest=false

## SERVER CERT E AUTH
securityMode=None
securityPolicy=None
userName=
password=

#securityMode=SIGNANDENCRYPT
#securityPolicy=1Basic256
#password=password1
#userName=user1

#api-ip=192.168.13.153

## ADMINISTRATION SERVICES
api-port=8080

## POLL COMMANDS SETTINGS
polling=false
polling-commands-timer=1000
pollingDaemonFrequency=20000
pollingExpiration=200000

## AGENT ID
agent-id=age01_
entity-id=age01_Car # used only during tests

## CONFIGURATION
configuration=#api

## CHECK TIMER POLLING DEVICES
checkTimer=2000
```
I set `namespaceIndex` `namespaceNumericIdentifier` to identify the parent object of all the variable, is that correct?

MattiaMarzano-Eng, 2022-05-27:
Dear @drypatrick, your configuration is correct. 
We're developing a new mapping tool, you can find it in the dedicated branch. In case of need, you can even try to run that version of the agent locally "only" to produce the config.json. Obviously keep in mind that is not released yet, then you can find some bugs.

drypatrick, 2022-05-27:
> Dear @drypatrick, your configuration is correct. We're developing a new mapping tool, you can find it in the dedicated branch. In case of need, you can even try to run that version of the agent [...]

It's good to know that the development of the tool is continuing. It would be interesting to have a debug function on the mapping tool, so that we can figure out why the mapping is being interrupted and remedy it server-side

drypatrick, 2022-05-27:
> Dear @drypatrick, your configuration is correct. We're developing a new mapping tool, you can find it in the dedicated branch. In case of need, you can even try to run that version of the agent [...]

I've tested the new mapping tool, but it still produces an empty file, it seems to ignore the variable present on the nameSpace 4.
The case could be that the variables in my PLC are identified as `ns=<nameSpace>;i=<identifier>` instead of  `ns=<nameSpace>;n=<name>`? In the following table, some variables on my server

Display   Name | Node Id | Value | Data Type
-- | -- | -- | --
SECOND | ns=4;i=34 | 9 | Byte
actAngleOffset | ns=4;i=132 | 674966 | Single
SECOND | ns=4;i=219 | 14 | Byte
rotationAngle | ns=4;i=42 | 674903 | Single
randomTimer | ns=4;i=230 | 3164 | Int32
NANOSECOND | ns=4;i=35 | 982669000 | UInt32
processStatus | ns=4;i=198 | In QC | String

Thanks for the support!!

MattiaMarzano-Eng, 2022-05-27:
It could be an issue related to the variable in your PLC. Unfortunately we haven't a wide range of "open" server to test our solution. My suggestion is to navigate your server using this [client](https://www.prosysopc.com/products/opc-ua-browser/) and look at our simple [logic](https://github.com/Engineering-Research-and-Development/iotagent-opcua/blob/mappingTool/mappingTool/objectFolder.js) that build the config.json. In case you find what is missing we can add the missing mapping in our tool. Otherwise the fastest option is to add manually your devices if the mapping tool can't do it.

closed 2022-12-27 by manfredipist