Engineering-Research-and-Development/iotagent-opcua #73: Docker-compose file

drypatrick, 2022-05-11:
In the docker-composer files there are some unclear comments, in particular [at the line](https://github.com/Engineering-Research-and-Development/iotagent-opcua/blob/c71ea2419346a11f6473fe0cde21811bcf03cb4e/docker-compose.yml#L35) is told to comment if you want to use NGSI-LD or to comment [the line](https://github.com/Engineering-Research-and-Development/iotagent-opcua/blob/c71ea2419346a11f6473fe0cde21811bcf03cb4e/docker-compose.yml#L37) if you want to use NGSI-V2.

Reading the strings that should be commented out however, it would seem that it is necessary to remove the comments from the lines to use NGSI-LD, and comment them if you want to use NGS-V2.

Is my interpretation correct? Thanks for clearing it up.

MattiaMarzano-Eng, 2022-05-20:
Hello, sorry for the unclear comments.
Your interpretation is correct. I'll fix those hints in the docker-compose files.
You should comment  out `- IOTA_CB_NGSI_VERSION=ld` if you want to use NGSI-LD
The same thing for orion-ld, in the following the correct setup:
`orion:
    hostname: orion 
    image: fiware/orion-ld:0.7.0
    networks:
      - hostnet
      - ocbnet
    ports:
      - "1026:1026"
    depends_on:
      - orion_mongo
    command: -statCounters -dbhost orion_mongo -logLevel INFO -forwarding
`

By default the Agent use NGSIv2, please apply those changes in case you want to use the agent with NGSI-LD.
Keep in mind to use a [config.json](https://github.com/Engineering-Research-and-Development/iotagent-opcua/blob/c71ea2419346a11f6473fe0cde21811bcf03cb4e/conf/config.json_NGSI-ld) like the example provided for NGSI-LD

closed 2022-05-20 by MattiaMarzano-Eng

drypatrick, 2022-05-27:
> Hello, sorry for the unclear comments. Your interpretation is correct. I'll fix those hints in the docker-compose files. You should comment out `- IOTA_CB_NGSI_VERSION=ld` if you want to use NGSI-LD [...]

Just a last question, [Jason Fox on stackoverflow said](https://stackoverflow.com/a/72272462/6224317) that also the `IOTA_FALLBACK_TENANT=opcua_car` is a requirement for NGSI-ld, is that correct? Should be commented using NGSI-v2?

Thanks!