Engineering-Research-and-Development/iotagent-opcua #75: OPC-UA iotagent4fiware/iotagent-opcua:latest NOT RUNNING (4/5 containers are running)

umerusman34, 2022-06-16:
I am going through the OPC UA Agent Tutorial
[image]

But after running the docker-compose one of the containers is not working and keeps giving errors. I have tried to delete every image and container and then run it again but still not working

[image]

[image]

[image]

MattiaMarzano-Eng, 2022-06-16:
Probably you have accidentally deleted the config.properties file from AGECONF directory. It is mandatory to run properly the agent, what you can delete is the config.json.

umerusman34, 2022-06-16:
I have it in the AGECONF, I have done nothing I am just using the docker-compose up after cloning it. I have tried it on another laptop but the same issue.
[## SOUTHBOUND CONFIGURATION (OPC UA)
namespace-ignore=2,7
endpoint=opc.tcp://iotcarsrv:5001/UA/CarServer

## NORTHBOUND CONFIGURATION (ORION CONTEXT BROKER)
context-broker-host=orion
context-broker-port=1026
fiware-service=opcua_car
fiware-service-path=/demo

## AGENT CONFIGURATION
server-base-root=/
server-port=4001
provider-url=http://iotage:4001

device-registration-duration=P1M
device-registry-type=memory

log-level=INFO

namespaceIndex=3
namespaceNumericIdentifier=1000

# MONGO-DB CONFIGURATION (required if device-registry-type=mongodb)
mongodb-host=iotmongo
mongodb-port=27017
mongodb-db=iotagent
mongodb-retries=5
mongodb-retry-time=5

## DATATYPE MAPPING OPCUA --> NGSI
OPC-datatype-Number=Number
OPC-datatype-Decimal128=Number
OPC-datatype-Double=Number
OPC-datatype-Float=Number
OPC-datatype-Integer=Integer
OPC-datatype-UInteger=Integer
OPC-datatype-String=Text
OPC-datatype-ByteString=Text
#END DATATYPE MAPPING OPCUA --> NGSI

## SESSION PARAMETERS
requestedPublishingInterval=10
requestedLifetimeCount=1000
requestedMaxKeepAliveCount=10
maxNotificationsPerPublish=100
publishingEnabled=true
priority=10

#SubscriptionsStrategy
uniqueSubscription=false

## MONITORING PARAMETERS
samplingInterval=1
queueSize=10000
discardOldest=false

## SERVER CERT E AUTH
securityMode=None
securityPolicy=None
userName=
password=

#securityMode=SIGNANDENCRYPT
#securityPolicy=1Basic256
#password=password1
#userName=user1

#api-ip=192.168.13.153

## ADMINISTRATION SERVICES
api-port=8080

## POLL COMMANDS SETTINGS
polling=false
polling-commands-timer=1000
pollingDaemonFrequency=20000
pollingExpiration=200000

## AGENT ID
agent-id=age01_
entity-id=age01_Car # used only during tests

## CONFIGURATION
configuration=#api

## CHECK TIMER POLLING DEVICES
checkTimer=2000
](url)

umerusman34, 2022-06-16:
@MattiaMarzano-Eng @drypatrick Can you try this tutorial at your end and check? It will not take much time
https://github.com/Engineering-Research-and-Development/iotagent-opcua/blob/master/docs/opc_ua_agent_tutorial.md

umerusman34, 2022-06-16:
[image]
It worked!
/AGECONF/config:/opt/iotagent-opcua/conf/config

I have added these bold words! Thank you for your help :)

closed 2022-06-16 by umerusman34