Engineering-Research-and-Development/iotagent-opcua #77: Sending Northbound traffic from IoTAgent OPCUA to Orion Context Broker

Anupamskd7, 2022-08-22:
Hi, according to the [doc](https://iotagent-opcua.readthedocs.io/en/latest/opc_ua_agent_tutorial/index.html), I have deployed OPC UA server, OPC UA Agent.
Now, following the doc, I provisioned the "car" device and updated the Orion entity for "accelerate" attribute.

What I want is to send device alerts (Northbound data) from IoT Agent which will update it on Orion CB. (Like send attribute to 7896 port in other IoTAgents).

Also, if this is done via the OPC UA Server, could you please share the procedure?

MattiaMarzano-Eng, 2022-08-23:
Hi,
if you need to send a kind of attribute that should be updated by a condition, from my perspective it should be implemented in the server and monitored by the agent, the other option could be send the update directly from the device (server) to the CB using the proper APIs. If I didn't get the point please elaborate a bit more what you have in mind to do, it's not totally clear for me.

Anupamskd7, 2022-08-29:
Hi @MattiaMarzano-Eng, What I want is a clear understanding of Northbound and Southbound data flow. I would really appreciate if you could please explain these two methods that you have mentioned in more detail, or provide some APIs.

What my current understanding of Northbound data flow (Car server example) is, After provisioning the car device:
`curl --location --request POST 'http://localhost:4001/iot/devices' \
--header 'fiware-service: opcua_car' \
--header 'fiware-servicepath: /demo' \
--header 'Content-Type: application/json' \
--data-raw '{
    "devices": [
        {
            "device_id": "age02_Car",
            "entity_name": "age02_Car",
            "entity_type": "Device",
            "attributes": [
                {
                    "object_id": "ns=3;s=EngineBrake",
                    "name": "EngineBrake",
                    "type": "Number"
                },
                {
                    "object_id": "ns=3;s=Acceleration",
                    "name": "Acceleration",
                    "type": "Number"
                },
                {
                    "object_id": "ns=3;s=EngineStopped",
                    "name": "EngineStopped",
                    "type": "Boolean"
                },
                {
                    "object_id": "ns=3;s=Temperature",
                    "name": "Temperature",
                    "type": "Number"
                },
                {
                    "object_id": "ns=3;s=Oxigen",
                    "name": "Oxigen",
                    "type": "Number"
                }
            ],
            "lazy": [{
                    "object_id": "ns=3;s=Speed",
                    "name": "Speed",
                    "type": "Number"
                }
            ],
            "commands" : [{
                "object_id": "ns=3;s=Stop",
                "name" : "Stop",
                "type" : "command"
            }, {
                "object_id": "ns=3;s=Accelerate",
                "name" : "Accelerate",
                "type" : "command"
            }]
      }
    ]
}
'`
Send an update to the Orion CB to Accelerate the car:
`curl --location --request PUT 'http://localhost:1026/v2/entities/age02_Car/attrs/Accelerate?type=Device' \
--header 'content-type: application/json' \
--header 'fiware-service: opcua_car' \
--header 'fiware-servicepath: /demo' \
--data-raw '{
  "value": [1],
  "type": "command"
}'`

and then the Car server will update the EngineStopped attribute to False. (Northbound data flow of EngineStopped attribute from car server to IoTAgent to Orion CB).

Should I consider this understanding of Northbound flow correct and complete?
Is the Car Server updating the EngineStopped value?

MattiaMarzano-Eng, 2022-08-29:
Hello @Anupamskd7,
the Device you're provisioning is simulated by the OPC UA server provided in the demo (southbound), then the IoT Agent monitors the status of the attributes (Oxigen, Speed,...) and methods/commands (Accelerate, Stop, ..) putting them at the disposal of CB in NGSI format (Northbound) for each change coming from the server. 

When you send the update to CB to Accelerate the car, the Agent is able to share that information to the server which will update the EngineStopped Attribute in False as you well understood (again through the agent in charge of update the values on the CB)

closed 2022-08-29 by MattiaMarzano-Eng