Engineering-Research-and-Development/iotagent-opcua #81: Tutorial not fully working

prahvin, 2022-09-08:
Ref:- https://github.com/Engineering-Research-and-Development/iotagent-opcua/issues/79 

Tried after the fix for the above issue, works partially now.

This request works 
curl -X PUT 'http://localhost:1026/v2/entities/age01_Car/attrs/Accelerate?type=Device' -H 'content-type: application/json' -H 'fiware-service: opcua_car' -H 'fiware-servicepath: /demo' -d '{ "value": [2], "type": "command" }'

but the GET request as shown below still reports error
curl -X GET http://localhost:1026/v2/entities/age01_Car/attrs/Speed -H 'fiware-service: opcua_car' -H 'fiware-servicepath: /demo' {"error":"NotFound","description":"The entity does not have such an attribute"}

MattiaMarzano-Eng, 2022-09-20:
Please try to follow this [readme](https://github.com/Engineering-Research-and-Development/iotagent-opcua/tree/FIWARE_Summit/demo), keeping the config.propreties as is in the master (without changing endpoint property )

prahvin, 2022-11-29:
Fixed via PR https://github.com/Engineering-Research-and-Development/iotagent-opcua/pull/86

closed 2022-11-29 by prahvin