Engineering-Research-and-Development/iotagent-opcua #82: Support of Scorpio Context Broker?

IngMiad, 2022-10-18:
Hello,

I have a question regarding the support of Scorpio Context Broker. Actually it seems that iotagent-opcua only supports Orion Context Broker - is it planned to also support Scorpio in the future?

Greetings!

MattiaMarzano-Eng, 2022-10-18:
Hello,
actually we're already compliant with NGSI-ld. You can try to setup (if you haven't done it yet)
```
 "contextBroker": {
		"host": "scorpioHostname",
		"port": scorpioPort
	}
```
in config.js to forward the requests to Scorpio Context Broker.
Never had experience with that, but my expectation is that we're ready to use that CB since we support NGSI-ld

rosieks, 2022-10-19:
Actually I'm also trying to work with Scorpio. What about different URLs. I think that right now agent call `/v2/entities` while scorpio/ngsi-ld assumes it will be `ngsi-ld/v1/entities`

MattiaMarzano-Eng, 2022-10-19:
You can solve that issue using the [config.json](https://github.com/Engineering-Research-and-Development/iotagent-opcua/blob/master/conf/config.json_NGSI-ld) template provided as example where you can find the additional fields 
```
"ngsiVersion": "ld",
"jsonLdContext": "https://uri.etsi.org/ngsi-ld/v1/ngsi-ld-core-context.jsonld",
```
allowing you to call `ngsi-ld/v1/entities` enpoint

IngMiad, 2022-10-20:
Hello,

thank you for the quick response. Yes after some experiments it seems to work also with scorpio. I thought that registration URLs might differ in case of Scorpio, but this is not the case.

The differences I recognized is that it is hard to use mapping tool and auto-configuration. Scorpio needs much more time to start than Orion context broker, so that the automatic registrations at context broker fail (at least in my case). I tried to use docker-compose 'depends_on', but Scorpio needs around 20-30 seconds to 'really start', so that the IoT-Agent can't connect to broker for device registration. 

So you have to add a wait-for-scorpio-to-be-ready mechanism or use REST API for device/registration configuration.

closed 2022-10-20 by IngMiad

Anupamskd7, 2022-11-03:
Hi,

I tried as suggested by @MattiaMarzano-Eng by making necessary changes in the conf/config.json:

`{
"logLevel" : "INFO",
"multiCore" : true,
"relaxTemplateValidation":true,
"contextBroker" : {
  "host" : "gateway.default.svc.cluster.local",
  "port" : 9090
},
"server" : {
  "port" : 4001,
  "baseRoot" : "/"
},
"deviceRegistry" : {
  "type" : "memory"
},
"mongodb" : {
  "host" : "iotmongo",
  "port" : "27017",
  "db" : "iotagent_opcua",
  "retries" : 5,
  "retryTime" : 5
},
"types" : {
  "Device" : {
    "service" : "opcua_car",
    "subservice" : "/demo",
    "active" : [{
      "name" : "EngineBrake",
      "type" : "Number"
    }, {
      "name" : "Acceleration",
      "type" : "Number"
    }, {
      "name" : "EngineStopped",
      "type" : "Boolean"
    }, {
      "name" : "Engine_Temperature",
      "type" : "Number"
    }, {
      "name" : "Engine_Oxigen",
      "type" : "Number"
    }, {
      "name" : "DataBlocksGlobal_3_dbRfidCntr_3_ID1_3_xBusy",
      "type" : "String"
    }, {
      "name" : "DataBlocksGlobal_3_dbRfidCntr_3_ID1_3_xBusyStatus",
      "type" : "Boolean"
    }, {
      "name" : "DataBlocksGlobal_3_dbRfidCntr_3_ID1_3_xDone",
      "type" : "String"
    }, {
      "name" : "DataBlocksGlobal_3_dbRfidCntr_3_ID1_3_xDoneStatus",
      "type" : "Boolean"
    }],
    "lazy" : [{
        "name": "Speed",
        "type": "Number"
    }],
    "commands" : [{
      "name" : "Error",
      "type" : "command"
    }, {
      "name" : "Stop",
      "type" : "command"
    }, {
      "name" : "Accelerate",
[... 178 lines cut ...]
`"url": "http://orion:1026/ngsi-ld/v1/entityOperations/upsert/?options=update",
    "method": "POST",
    "headers": {
        "fiware-service": "opcua_car",
        "fiware-servicepath": "/demo",
        "Content-Type": "application/ld+json",
        "NGSILD-Tenant": "opcua_car",
        "NGSILD-Path": "/demo"
    },
    "json": [
        {
            "@context": "https://uri.etsi.org/ngsi-ld/v1/ngsi-ld-core-context-v1.3.jsonld",
            "DataBlocksGlobal_3_dbRfidCntr_3_ID1_3_xDoneStatus": {
                "type": "Property",
                "value": true
            },
            "id": "urn:ngsi-ld:Device:age01_Car",
            "type": "Device"
        }
    ]
}`

And the default service that gets provisioned also has **_cbHost": "orion:1026_**

@MattiaMarzano-Eng Could you please tell which config files are needed to be changed?

MattiaMarzano-Eng, 2022-11-04:
Hello @Anupamskd7, please look at this [template ](https://github.com/Engineering-Research-and-Development/iotagent-opcua/blob/master/conf/config.json_NGSI-ld)to update accordingly your config.json where are missing those fields `"ngsiVersion": "ld",
"jsonLdContext": "https://uri.etsi.org/ngsi-ld/v1/ngsi-ld-core-context.jsonld",`

update also the docker compose 
`
#################################################################################################################
      # please comment out if you want to use NGSI-ld, the OPCUA Agent is provided with NGSIv2 as default configuration
      #- IOTA_CB_NGSI_VERSION=ld
      #- IOTA_JSON_LD_CONTEXT=https://uri.etsi.org/ngsi-ld/v1/ngsi-ld-core-context-v1.3.jsonld
      #################################################################################################################
`
in thi section