Engineering-Research-and-Development/iotagent-opcua #84: Attributes are not updated in the OCB

alberto072710, 2022-11-22:
I am trying to communicate with an external opc server, the communication is established correctly and the iot agent receives the updates and sends the patch command to the OCB but it does not update the values of the attributes.

[image]

The OCBreceives the patch command

[image]

But when I query via postman the attribute has not been updated

[image]

If I send the same patch command as the OCB agent through postman if it updates the attribute value

[image]
[image]
[image]

Thank you very much for the help.

manfredipist, 2022-11-22:
Hello @alberto072710, I need few more informations to support you.
1) what version of iotagent are you using? Could you repeat this exact scenario on iotagent v2? you can find it both on dockerhub (iotagent4fiware/iotagent-opcua:2.0.0) and under the "iotagent-v2" branch
2) Could you provide your config.js configuration once you've customized the examples provided?
 
P.S You should be using the PUT method to update the value of any of your entities' attributes instead of overwriting them with the PATCH 
e.g:
`curl --location --request PUT 'http://localhost:1026/v2/entities/age01_Car/attrs/Engine_Oxigen?type=Device' \
--header 'Accept: application/json' \
--header 'Fiware-Service: opcua_car' \
--header 'Fiware-ServicePath: /demo' \
--header 'Content-Type: application/json' \
--data-raw '{
    "value": 5,
    "type":"Number"
}'`

alberto072710, 2022-11-22:
I have tried to do it with iotagent4fiware/iotagent-opcua:2.0.0 but any logs appear...

The config.js.

`{
  "logLevel": "DEBUG",
  "contextBroker": {
    "host": "orion",
    "port": 1026
  },
  "server": {
    "port": 4001,
    "baseRoot": "/"
  },
  "deviceRegistry": {
    "type": "memory"
  },
  "mongodb": {
    "host": "iotmongo",
    "port": "27017",
    "db": "iotagent",
    "retries": 5,
    "retryTime": 5
  },
  "types": {
    "Habitacion": {
      "service": "opcua_car",
      "subservice": "/demo",
      "active": [
        {
          "name": "temperature",
          "type": "Float"
        },
        {
          "name": "pressure",
          "type": "Integer"
        }
      ],
      "lazy": [],
      "commands": []
    }
  },
  "browseServerOptions": null,
  "service": "opcua_car",
  "subservice": "/demo",
  "providerUrl": "http://iotage:4001",
  "pollingExpiration": "200000",
  "pollingDaemonFrequency": "20000",
  "deviceRegistrationDuration": "P1M",
  "defaultType": null,
  "contexts": [
    {
      "id": "Habitacion:001",
      "type": "Habitacion",
      "service": "opcua_car",
      "subservice": "/demo",
      "polling": "false",
      "mappings": [
        {
          "ocb_id": "pressure",
          "opcua_id": "ns=4;i=4",
          "object_id": null,
          "inputArguments": []
        },
        {
          "ocb_id": "temperature",
          "opcua_id": "ns=4;i=3",
          "object_id": null,
          "inputArguments": []
        }
      ]
    }
  ],
  "contextSubscriptions": []
}`

The docker-compose file:

version: "3"
secrets:
   age_idm_auth:
      file: age_idm_auth.txt

services:
  iotage:
    hostname: iotage
    image: iotagent4fiware/iotagent-opcua:latest
    networks:
      - hostnet
      - iotnet
    ports:
      - "4001:4001"
      - "4081:8080"
    depends_on:
      - iotmongo
      - orion
    volumes:
      - ./AGECONF:/opt/iotagent-opcua/conf
    command: /usr/bin/tail -f /var/log/lastlog
    extra_hosts:
      - "iotcarsrv:192.168.2.1"
      - "SIMATIC.S7-1200:192.168.2.1"

  iotmongo:
    hostname: iotmongo
    image: mongo:4.4
    networks:
      - iotnet
    volumes:
      - iotmongo_data:/data/db
      - iotmongo_conf:/data/configdb

  ################ OCB ################

  orion:
    hostname: orion
    image: fiware/orion:latest
    networks:
      - hostnet
      - ocbnet
    ports:
      - "1026:1026"
    depends_on:
      - orion_mongo
    #command: -dbhost mongo
    entrypoint: /usr/bin/contextBroker -fg -multiservice -ngsiv1Autocast -statCounters -dbhost mongo -logForHumans -logLevel DEBUG -t 255

  orion_mongo:
    hostname: orion_mongo
    image: mongo:4.4
    networks:
      ocbnet:
        aliases:
          - mongo
    volumes:
      - orion_mongo_data:/data/db
      - orion_mongo_conf:/data/configdb
    command: --nojournal

volumes:
  iotmongo_data:
  iotmongo_conf:
  orion_mongo_data:
  orion_mongo_conf:

networks:
  hostnet:
  iotnet:
  ocbnet:

manfredipist, 2022-11-22:
IotAgent v2.0 brings many breaking changes thus you should re-download the whole project from here: https://github.com/Engineering-Research-and-Development/iotagent-opcua/tree/iotagent-v2 and customize the config.js file you can find in "conf" folder (examples are also provided)

Once you've configured your agent, you can run it locally by executing, from the root of the project, the "iotagent-opcua" binary you can find in the "bin" folder, otherwise, you can run it with docker by running the "docker-compose up" command from the "docker" folder. P.S: Don't forget to edit the docker-compose according to your needs.

alberto072710, 2022-11-23:
I just tried version 2 and it still doesn't update the attributes. I attach the screenshots where it is seen that the iot agent receives the temperature value 27 and the screenshot where the fiware-orion receives that 27. When I do a get of the entities from postman I receive another value.

[image]
[image]
[image]

```
var config = {};

config.iota = {
    /**
     * Configures the log level. Appropriate values are: FATAL, ERROR, INFO, WARN and DEBUG.
     */
    logLevel: 'DEBUG',
    /**
     * When this flag is active, the IoTAgent will add the TimeInstant attribute to every entity created, as well
     * as a TimeInstant metadata to each attribute, with the current timestamp.
     */
    timestamp: true,
    /**
     * Context Broker configuration. Defines the connection information to the instance of the Context Broker where
     * the IoT Agent will send the device data.
     */
    contextBroker: {
        /**
         * Host where the Context Broker is located.
         */
        host: 'localhost',
        /**
         * Port where the Context Broker is listening.
         */
        port: '1026',
        /**
         * Version of the Context Broker (v2 or ld)
         */
        ngsiVersion: 'v2',
        /**
         * JSON LD Context
         */
        jsonLdContext: 'https://uri.etsi.org/ngsi-ld/v1/ngsi-ld-core-context.jsonld',
        /**
         * Used as fallbackTenant
         */
        service: 'opcua_car',
        /**
         * Used as fallbackPath
         */
        subservice: '/demo'
    },
    /**
     * Configuration of the North Port of the IoT Agent.
     */
    server: {
        /**
         * Port where the IoT Agent will be listening for NGSI and Provisioning requests.
         */
        port: 4041
    },

    /**
     * Configuration for secured access to instances of the Context Broker secured with a PEP Proxy.
     * For the authentication mechanism to work, the authentication attribute in the configuration has to be fully
     * configured, and the authentication.enabled subattribute should have the value `true`.
     *
     * The Username and password should be considered as sensitive data and should not be stored in plaintext.
     * Either encrypt the config and decrypt when initializing the instance or use environment variables secured by
     * docker secrets.
     */
    //authentication: {
    //enabled: false,
[... 204 lines cut ...]
 * flag indicating whether the incoming notifications to the IoTAgent should be processed using the bidirectionality
 * plugin from the latest versions of the library or the OPCUA-specific configuration retrieval mechanism.
 */
config.configRetrieval = false;
/**
 * Default API Key, to use with device that have been provisioned without a Configuration Group.
 */
config.defaultKey = 'iot';
/**
 * Default transport protocol when no transport is provisioned through the Device Provisioning API.
 */
config.defaultTransport = 'OPCUA';
/**
 * flag indicating whether the node server will be executed in multi-core option (true) or it will be a
 * single-thread one (false).
 */
//config.multiCore = false;
/**
 * flag indicating whether or not to provision the Group and Device automatically
 */
config.autoprovision = true;

module.exports = config;
```
Thanks.

manfredipist, 2022-11-23:
Hello @alberto072710 , could you please try to test the environment with the following configs?

docker-compose.yml
```docker-compose.yml
version: "3.1"

volumes:
  mongodb: ~

services:
  iot-agent:
    image: iotagent4fiware/iotagent-opcua:2.0.0
    hostname: iotagent-opcua
    depends_on:
      - mongodb
      - orion
    networks:
      - hostnet
    ports:
      - "4041:4041"
      - "9229:9229"
    environment:
      - "IOTA_CB_HOST=orion"
      - "IOTA_CB_PORT=1026"
      - "IOTA_NORTH_PORT=4041"
      - "IOTA_REGISTRY_TYPE=mongodb"
      - "IOTA_MONGO_HOST=mongodb"
      - "IOTA_MONGO_PORT=27017"
      - "IOTA_MONGO_DB=iotagent_opcua"
      - "IOTA_PROVIDER_URL=http://iotagent-opcua:4041"
    volumes:
      - ../conf/config.js:/opt/iotagent-opcua/conf/config.js
    extra_hosts:
      - "SIMATIC.S7-1200:192.168.2.1"

  mongodb:
    image: mongo:4.2
    hostname: mongodb
    networks:
      - hostnet
    expose:
      - "27017"
    command: --bind_ip_all
    volumes:
      - mongodb:/data

  orion:
    image: fiware/orion
    hostname: orion
    depends_on:
      - mongodb
    networks:
      - hostnet
    ports:
      - "1026:1026"
    command: -dbhost mongodb -logLevel DEBUG

networks:
  hostnet:
```

config.js
```config.js
var config = {};
config.iota = {
    logLevel: 'DEBUG',
    timestamp: true,
    contextBroker: {
        host: 'orion',
        port: '1026',
        ngsiVersion: 'v2',
        jsonLdContext: 'https://uri.etsi.org/ngsi-ld/v1/ngsi-ld-core-context.jsonld',
        service: 'opcua_home',
        subservice: '/demo'
    },
    server: {
        port: 4041
    },
    deviceRegistry: {
        type: 'mongodb'
    },
    mongodb: {
        host: 'mongodb',
        port: '27017',
        db: 'iotagent_opcua'
    },
    types: {
        Habitacion: {
            active: [
                {
                    name: 'pressure',
                    type: 'Integer'
                },
                {
                    name: 'temperature',
                    type: 'Float'
                }
            ],
            lazy: [],
            commands: []
        }
    },
    contexts: [
        {
            id: 'Habitacion:001',
            type: 'Habitacion',
            mappings: [
                {
			"ocb_id": "pressure",
			"opcua_id": "ns=4;i=4",
			"object_id": null,
			"inputArguments": []
		},
		{
			"ocb_id": "temperature",
			"opcua_id": "ns=4;i=3",
			"object_id": null,
			"inputArguments": []
		}
            ]
        }
    ],
    contextSubscriptions: [],
    service: 'opcua_home',
    subservice: '/demo',
    providerUrl: 'http://iotagent-opcua:4041',
    deviceRegistrationDuration: 'P20Y',
    defaultType: 'Habitacion',
    defaultResource: '/iot/opcua',
    explicitAttrs: false
};

config.opcua = {
    endpoint: 'opc.tcp://192.168.2.1:4840/PLC',
    securityMode: 'None',
    securityPolicy: 'None',
    username: null,
    password: null,
    uniqueSubscription: false,
    polling: false,
    agentId: 'age01_',
    namespaceIgnore: '0,7',
    entityId: 'Habitacion:001',
    entityType: 'Habitacion'
};

config.jexlTransformations = {};

config.configRetrieval = false;
config.defaultKey = 'iot';
config.defaultTransport = 'OPCUA';
//config.multiCore = false;
config.autoprovision = true;

module.exports = config;
```

alberto072710, 2022-11-23:
It finally worked. I forgot to put the header when doing the get

```
curl -X GET \
  http://localhost:1026/v2/entities \
  -H 'fiware-service: opcua_car' \
  -H 'fiware-servicepath: /demo'
```

manfredipist, 2022-11-23:
Glad to hear that. Closing the issue.

closed 2022-11-23 by manfredipist