Engineering-Research-and-Development/iotagent-opcua #85: Write commands/methods error

alberto072710, 2022-11-24:
I'm trying to execute methods on the opc server but I can't. I have the following method in my opc server.

[image]

My config.js
```
var config = {};

config.iota = {
    logLevel: 'DEBUG',
    timestamp: true,
    contextBroker: {
        host: 'localhost',
        port: '1026',
        ngsiVersion: 'v2',
        jsonLdContext: 'https://uri.etsi.org/ngsi-ld/v1/ngsi-ld-core-context.jsonld',
        service: 'opcua_car',
        subservice: '/demo'
    },
    server: {
        port: 4041
    },
    deviceRegistry: {
        type: 'mongodb'
    },
    mongodb: {
        host: 'localhost',
        port: '27017',
        db: 'iotagent_opcua'
    },
    types: {
		Habitacion : {
			  active : [ {
				name : "temperature",
				type : "Float"
			  }, {
				name : "pressure",
				type : "Integer"
			  } ],
			  lazy : [ ],
			  commands : [{
				  name : "temperatureCommand",
				  type : "command"
			  }]
		}
	},
    contexts: [
	{
		id: "Habitacion:001",
		type: "Habitacion",
		polling: "false",
		mappings: [
			{
				ocb_id: "pressure",
				opcua_id: "ns=4;i=4",
				object_id: null,
				inputArguments: []
			},
			{
				ocb_id: "temperature",
				opcua_id: "ns=4;i=3",
				object_id: null,
				inputArguments: []
			}
		]
	}
	],
    contextSubscriptions: [{
		id: "Habitacion:001",
		type: "Habitacion",
		mappings: [
			{
				ocb_id: "temperatureCommand",
				opcua_id: "ns=4;i=9",
				object_id: "ns=4;i=9",
				inputArguments: []
	}]
	}],
    service: 'opcua_car',
    subservice: '/demo',
    providerUrl: 'http://localhost:4041',
    deviceRegistrationDuration: 'P20Y',
    defaultType: 'Device',
    defaultResource: '/iot/opcua',
    explicitAttrs: false
};

config.opcua = {
    endpoint: 'opc.tcp://192.168.2.1:4840/PLC',
    securityMode: 'None',
    securityPolicy: 'None',
    username: null,
    password: null,
    uniqueSubscription: false,
    polling: false,
    agentId: 'age01_',
    namespaceIgnore: '0,7',
    entityId: 'age01_Car',
    entityType: 'Device'
};
config.jexlTransformations = {};
config.configRetrieval = false;
config.defaultKey = 'iot';
config.defaultTransport = 'OPCUA';
//config.multiCore = false;
config.autoprovision = true;
module.exports = config;
```
When I execute the command to activate the method it gives an error
[image]

But the iot-agent receives the update command

[image]

iot agent devices

[image]

ocb entities

[image]

walterwootz, 2022-11-24:
Hi, could you show the registrations on Orion by running the following request?
```
curl --location --request GET 'http://localhost:1026/v2/registrations' \
--header 'Accept: application/json' \
--header 'Fiware-Service: opcua_car' \
--header 'Fiware-ServicePath: /demo'
```

alberto072710, 2022-11-24:
[image]

walterwootz, 2022-11-29:
Hello @alberto072710, we fixed the problem with the new version of the agent 2.0.2, so please use image `iotagent4fiware/iotagent-opcua:2.0.2` on your docker compose and use the latest version of the config.js  you can find [here](https://github.com/Engineering-Research-and-Development/iotagent-opcua/tree/iotagent-v2/conf) (there are some small differences in the opcua/mappingTool section).

alberto072710, 2022-11-30:
Thank you very much. Amazing work!!!!

Now if I manage to send a command and I see how that command updates the corresponding variable, but the "info" entity does not get any value and the "status" entity of the command remains "pending", could you explain to me how these entities work? From an opc browser if I get success.

[image]

[image]

On the other hand, I have tried to declare a command "suma" with input parameters, but when executing them I get an error in the iot-agent, I attach the configuration in case there is something wrong declared

config.js

```
var config = {};

config.iota = {
    logLevel: 'DEBUG',
    timestamp: true,
    contextBroker: {
        host: 'localhost',
        port: '1026',
        ngsiVersion: 'v2',
        jsonLdContext: 'https://uri.etsi.org/ngsi-ld/v1/ngsi-ld-core-context.jsonld',
        service: 'opcua_car',
        subservice: '/demo'
    },
    server: {
        port: 4041
    },
    deviceRegistry: {
        type: 'mongodb'
    },
    mongodb: {
        host: 'localhost',
        port: '27017',
        db: 'iotagent_opcua'
    },
    types: {
        Habitacion: {
            active: [
                {
                    name: 'temperature',
                    type: 'Float'
                },
                {
                    name: 'pressure',
                    type: 'Integer'
                }
            ],
            lazy: [],
            commands: [
                {
                    name: 'temperatureCommand',
                    type: 'command'
                },
				{
					name: 'suma',
					type: 'command'
				}
            ]
        }
    },
    contexts: [
        {
            id: 'Habitacion:001',
            type: 'Habitacion',
            mappings: [
                {
                    ocb_id: 'pressure',
                    opcua_id: 'ns=4;i=4',
[... 211 lines cut ...]
2022-11-30 09:46:53         "suma_status": {
2022-11-30 09:46:53             "type": "commandStatus",
2022-11-30 09:46:53             "value": "PENDING",
2022-11-30 09:46:53             "metadata": {
2022-11-30 09:46:53                 "TimeInstant": {
2022-11-30 09:46:53                     "type": "DateTime",
2022-11-30 09:46:53                     "value": "2022-11-30T08:46:53.070Z"
2022-11-30 09:46:53                 }
2022-11-30 09:46:53             }
2022-11-30 09:46:53         },
2022-11-30 09:46:53         "TimeInstant": {
2022-11-30 09:46:53             "type": "DateTime",
2022-11-30 09:46:53             "value": "2022-11-30T08:46:53.070Z"
2022-11-30 09:46:53         }
2022-11-30 09:46:53     }
2022-11-30 09:46:53 } | comp=IoTAgent
2022-11-30 09:46:53 time=2022-11-30T08:46:53.074Z | lvl=DEBUG | corr=db0310f6-3ed8-4cd6-b130-2f95f929f2f7 | trans=db0310f6-3ed8-4cd6-b130-2f95f929f2f7 | op=IoTAgentNGSI.Request | from=n/a | srv=opcua_car | subsrv=/demo | msg=Response "" | comp=IoTAgent
2022-11-30 09:46:53 time=2022-11-30T08:46:53.074Z | lvl=INFO | corr=db0310f6-3ed8-4cd6-b130-2f95f929f2f7 | trans=db0310f6-3ed8-4cd6-b130-2f95f929f2f7 | op=IoTAgentNGSI.Entities-v2 | from=n/a | srv=opcua_car | subsrv=/demo | msg=Received the following response from the CB: Value updated successfully
2022-11-30 09:46:53  | comp=IoTAgent
2022-11-30 09:46:53 time=2022-11-30T08:46:53.074Z | lvl=DEBUG | corr=db0310f6-3ed8-4cd6-b130-2f95f929f2f7 | trans=db0310f6-3ed8-4cd6-b130-2f95f929f2f7 | op=IoTAgentNGSI.ContextServer-v2 | from=n/a | srv=opcua_car | subsrv=/demo | msg=Update action from [iotagent-opcua:4041] handled successfully. | comp=IoTAgent
2022-11-30 09:46:53 time=2022-11-30T08:46:53.074Z | lvl=DEBUG | corr=89a7f2aa-708b-11ed-806f-0242ac160003; cbfwd=1 | trans=203823ff-5b8e-42a0-8f31-7eb867a715bf | op=IoTAgentNGSI.DomainControl | from=n/a | srv=opcua_car | subsrv=/demo | msg=response-time: 12 | comp=IoTAgent
2022-11-30 09:46:54 time=2022-11-30T08:46:54.348Z | lvl=DEBUG | corr=4013beea-9c1a-4969-8ffb-b7974f807b49 | trans=4013beea-9c1a-4969-8ffb-b7974f807b49 | op=IoTAgentNGSI.GenericMiddlewares | from=n/a | srv=n/a | subsrv=n/a | msg=Request for path [/iot/about] from [localhost:4041] | comp=IoTAgent
2022-11-30 09:46:54 time=2022-11-30T08:46:54.348Z | lvl=DEBUG | corr=4013beea-9c1a-4969-8ffb-b7974f807b49 | trans=4013beea-9c1a-4969-8ffb-b7974f807b49 | op=IoTAgentNGSI.DomainControl | from=n/a | srv=n/a | subsrv=n/a | msg=response-time: 0 | comp=IoTAgent

```

walterwootz, 2022-12-02:
Hello @alberto072710, the problem has been fixed with the new version of the agent 2.0.3, so please use image iotagent4fiware/iotagent-opcua:2.0.3 on your docker compose.

If you encounter new problems, please open a new issue.

closed 2022-12-02 by walterwootz