Engineering-Research-and-Development/iotagent-opcua #89: Complex mapping

rosieks, 2022-12-21:
How can I map OPC attribute to more complex json property?

For example I would like to have model like that:

```
{
   "id": "some-id",
   "type": "Sensor",
   "hasFunction": {
       "type": "property",
       "value": {
           "hasSensingRange": {
              "type": "Property",
              "observedAt": "2022-12-02T01:48:29Z", <--- this one I know is going to be introduced in next version
              "value": 23, <---- this one is taken from another NodeId
              "unitCode": "°C" <---- this one is taken from NodeId (nested inside previous one)
           },
       }
   }
}
```

Ideally it would be nice to have more detailed documentation for mapping

manfredipist, 2022-12-21:
Hi @rosieks ! Could you please provide a working example of your OPC UA server?

rosieks, 2022-12-21:
What do you mean by working example? What I would like to achieve is that value is not scalar value mapped from single node but rather complex object/type composed from multiple nodes in OPC

manfredipist, 2022-12-21:
I meant if you've got an opcua server we could instantiate on our own, or access to test if we can support this kind of feature

rosieks, 2022-12-21:
Unfortunatelly OPCUA server that I use is in private network and is connected to real building. But this issue is not OPC server specific. It's just matter how do you model your context

rosieks, 2022-12-29:
But good example for that would be Engine from the sample here. I don't want to have two attributes `Engine_Temperature` and `Engine_Oxigen`. Instead I want to have one attribute `Engine` and it will be object containing two attributes: `Temperature` and `Oxigen`. How can I map something like that?

manfredipist, 2023-01-12:
Hi @rosieks , looking at iotagent-lib test suite, i've found this particular example that exploits reverse expression to accompliash something like the one thing you've asked. Please have a look at it and let me if you make it!

https://github.com/telefonicaid/iotagent-node-lib/blob/5f2c1b78fa872d8a9d9486446790671b15f2afcf/test/unit/examples/deviceProvisioningRequests/provisionFullDevice.json

rosieks, 2023-01-17:
Hi @manfredipist - thanks for suggestion. After reading documentation it looks like first it requires extra plugin and second it's to solve problem of bidirectional mapping. So it doesn't solve the issue of mapping OPC node to this structure.

manfredipist, 2023-01-30:
Hi  @rosieks , it seems that the iotagent-node-lib needs flat-structured JSON provisioning and does not support complex or nested attributes. We suggest you create an issue first for the iotagent-node-lib and if it is (or will be) implemented, we will be happy to add support for the OPCUA IoT Agent as well. Regarding your case, you can monitor the "value" and "unitCode" nodes separately and then handle the aggregation logic outside the broker context.

closed 2023-02-21 by manfredipist