Engineering-Research-and-Development/iotagent-opcua #95: Connecting The OPC UA IoTAgent To An External OPC UA Server

drypatrick, 2023-01-17:
The [documentation ](https://iotagent-opcua.readthedocs.io/en/latest/user_and_programmers_manual/index.html#connecting-the-opc-ua-iotagent-to-an-external-opc-ua-server)seems to be out of date regarding the connection to an external server, the `docker-compose-external-server.yml` is missing from the repository, and it is not so clear how to proceed, is it possible to provide an example for remote configuration?

manfredipist, 2023-01-18:
Hi @drypatrick , I'm attaching the configuration for the docker-compose file of the new iotagent v2, in which **extra_host** property has been added. On you _config.js_ file you will need to refer to the opcua server accordingly, in this case by the hostname **_opcuaServerHostname_**

```yaml
version: "3.1"

volumes:
  mongodb: ~

services:
  iot-agent:
    image: iotagent4fiware/iotagent-opcua:latest
    hostname: iotagent-opcua
    depends_on:
      - mongodb
      - orion
    networks:
      - hostnet
    ports:
      - "4041:4041"
      - "9229:9229"
    extra_hosts:
      - "opcuaServerHostname:136.243.156.113"
    environment:
      - "IOTA_CB_HOST=orion"
      - "IOTA_CB_PORT=1026"
      - "IOTA_NORTH_PORT=4041"
      - "IOTA_REGISTRY_TYPE=mongodb"
      - "IOTA_MONGO_HOST=mongodb"
      - "IOTA_MONGO_PORT=27017"
      - "IOTA_MONGO_DB=iotagent_opcua"
      - "IOTA_PROVIDER_URL=http://iotagent-opcua:4041"
      - "IOTA_OPCUA_ENDPOINT=opc.tcp://iotcarsrv:5001/UA/CarServer"
      - "IOTA_OPCUA_SECURITY_MODE=None"
      - "IOTA_OPCUA_SECURITY_POLICY=None"
      #- "IOTA_OPCUA_SECURITY_USERNAME=null"
      #- "IOTA_OPCUA_SECURITY_PASSWORD=null"
      - "IOTA_OPCUA_UNIQUE_SUBSCRIPTION=false"
      - "IOTA_OPCUA_MT_POLLING=false"
      - "IOTA_OPCUA_MT_AGENT_ID=age01_"
      - "IOTA_OPCUA_MT_ENTITY_ID=age01_Car"
      - "IOTA_OPCUA_MT_ENTITY_TYPE=Device"
      - "IOTA_OPCUA_MT_NAMESPACE_IGNORE=0,7"
    volumes:
      - ../conf/config.js:/opt/iotagent-opcua/conf/config.js

  mongodb:
    image: mongo:4.2
    hostname: mongodb
    networks:
      - hostnet
    expose:
      - "27017"
    command: --bind_ip_all
    volumes:
      - mongodb:/data

  orion:
    image: fiware/orion
    hostname: orion
    depends_on:
      - mongodb
    networks:
      - hostnet
    ports:
      - "1026:1026"
    command: -dbhost mongodb -logLevel DEBUG

networks:
  hostnet:
```

drypatrick, 2023-01-18:
So all the entries of `opcuaServerHostname` will be overridden by `136.243.156.113`, ins't it?
Would be nice to configure also the value of the port and the subfolder getting the endpoint address from the docker-compose file

drypatrick, 2023-01-18:
The same thing applies to other variables that are set in both the `docker-compose` and in the `config.js`, for example, the IOTA_OPCUA_MT_NAMESPACE_IGNORE, IOTA_OPCUA_MT_ENTITY_TYPE, IOTA_OPCUA_MT_ENTITY_ID, etc.

Couldn't these be set only in the `docker-compose` file? Also because it seems to me that it only takes into account the values set in the `docker-compose` file

manfredipist, 2023-01-19:
Hi @drypatrick , sorry if i gave for granted that docker-compose envs override config.js file which is the entry point for configuring the agent. It's up to you if configuring the config.js and mapping into a volume OR configuring the docker-compose envs leaving the config.js as per default

drypatrick, 2023-01-19:
Ok, so I could edit the IOTA_OPCUA_ENDPOINT, without even add the extra host, if this entry should override the value in the config.js?

manfredipist, 2023-01-19:
It depends on various factors, your network, server's reachability, vpn requirements and so on.
Generally speaking if the server you want to reach is outside your localhost network you should need to set the extra_host property.