Engineering-Research-and-Development/iotagent-opcua #96: Multiple inputs command and output result

alberto072710, 2023-01-27:
Hi,
I am trying to send a command that sending two numbers gives me the sum of both as a result. 

I attach an image of the Prosys

[image]

The first question that arises is how can I send a command with multiple input arguments.

[image]

How are the output parameters received? What is the meaning of commandResult and commandStatus?

[image]

config.js

```
const config = {};

config.iota = {
    logLevel: 'DEBUG',
    timestamp: true,
    contextBroker: {
        host: 'localhost',
        port: '1026',
        ngsiVersion: 'v2',
        jsonLdContext: 'https://uri.etsi.org/ngsi-ld/v1/ngsi-ld-core-context.jsonld',
        service: 'opcua_car',
        subservice: '/demo'
    },
    server: {
        port: 4041
    },
    deviceRegistry: {
        type: 'mongodb'
    },
    mongodb: {
        host: 'localhost',
        port: '27017',
        db: 'iotagent_opcua'
    },
    types: {
        Habitacion: {
            active: [
                {
                    name: 'temperature',
                    type: 'Float'
                },
                {
                    name: 'pressure',
                    type: 'Integer'
                }
            ],
            lazy: [],
            commands: [
                {
                    name: 'temperatureCommand',
                    type: 'command'
                },
				{
					name: 'suma',
					type: 'command'
				},
				{
					name: 'ajusteTemperatura',
					type 'command'
				}
            ]
        }
    },
    contexts: [
        {
            id: 'Habitacion:001',
            type: 'Habitacion',
            mappings: [
                {
                    ocb_id: 'pressure',
                    opcua_id: 'ns=4;i=4',
                    object_id: 'ns=4;i=4',
                    inputArguments: []
                },
                {
                    ocb_id: 'temperature',
                    opcua_id: 'ns=4;i=3',
                    object_id: 'ns=4;i=3',
                    inputArguments: []
                }
            ]
        }
    ],
    contextSubscriptions: [
        {
            id: 'Habitacion:001',
            type: 'Habitacion',
            mappings: [
                {
                    ocb_id: 'temperatureCommand',
                    opcua_id: 'ns=4;i=8',
                    object_id: 'ns=4;i=1',
                    inputArguments: []
                },
				{
					ocb_id: 'suma',
                    opcua_id: 'ns=4;i=9',
                    object_id: 'ns=4;i=1',
                    inputArguments: [
						{
							dataType: 4,
							type: 'DATO1'
						},
						{
							dataType: 4,
							type: 'DATO2'
						}
					]	
				},
				{
					ocb_id: 'ajusteTemperatura',
                    opcua_id: 'ns=4;i=12',
                    object_id: 'ns=4;i=1',
                    inputArguments: [
						{
							dataType: 4,
							type: 'TEMPERATURA'
						}
					]	
				}
            ]
        }
    ],
	events:[],
    service: 'opcua_car',
    subservice: '/demo',
    providerUrl: 'http://localhost:4041',
    deviceRegistrationDuration: 'P20Y',
    defaultType: 'Device',
    defaultResource: '/iot/opcua',
    explicitAttrs: false
};

config.opcua = {
	subscription: {
		maxNotificationsPerPublish: 1000,
		publishingEnabled: true,
		requestedLifetimeCount: 100,
		requestedMaxKeepAliveCount: 10,
		requestedPublishingInterval: 1000,
		priority: 128
    },
    endpoint: 'opc.tcp://192.168.2.1:4840',
    securityMode: 'None',
    securityPolicy: 'None',
    username: null,
    uniqueSubscription: false
};

config.mappingTool = {
    polling: false,
    /**
     * agentId prefix to be assigned to the newly generated entity from MappingTool execution
     */
    agentId: 'age01_',
    /**
     * Namespaces to ignore when crawling nodes from OPC UA Server
     */
    namespaceIgnore: '0,7',
    /**
     * entityId to be assigned to the newly generated entity from MappingTool execution
     */
    entityId: 'age01_Car',
    /**
     * entityType to be assigned to the newly generated entity from MappingTool execution
     */
    entityType: 'Device'
};
config.jexlTransformations = {};
config.configRetrieval = false;
config.defaultKey = 'iot';
config.defaultTransport = 'OPCUA';
//config.multiCore = false;
config.autoprovision = true;
module.exports = config;
```
Thank you very much for your help

alberto072710, 2023-01-31:
Hi,
Thank you very much for your quick response.
I have tried what you told me but I have not been successful.
I run the command:
[image]

But I can't get the sum in the info attribute and the status attribute stays pending

[image]

If I run this command from prosys if I get the correct output.

[image]

The logs are as follows:

```
2023-01-31 13:35:25 time=2023-01-31T12:35:25.736Z | lvl=DEBUG | corr=bca84fd4-a163-11ed-bd94-0242ac140003; cbfwd=1 | trans=32ac83b6-1f8c-45d2-ba55-319dee4f7e16 | op=IoTAgentNGSI.GenericMiddlewares | from=n/a | srv=opcua_car | subsrv=/demo | msg=Request for path [//op/update] from [iotagent-opcua:4041] | comp=IoTAgent
2023-01-31 13:35:25 time=2023-01-31T12:35:25.736Z | lvl=DEBUG | corr=bca84fd4-a163-11ed-bd94-0242ac140003; cbfwd=1 | trans=32ac83b6-1f8c-45d2-ba55-319dee4f7e16 | op=IoTAgentNGSI.GenericMiddlewares | from=n/a | srv=opcua_car | subsrv=/demo | msg=Body:
2023-01-31 13:35:25 
2023-01-31 13:35:25 {
2023-01-31 13:35:25     "entities": [
2023-01-31 13:35:25         {
2023-01-31 13:35:25             "id": "Habitacion:001",
2023-01-31 13:35:25             "type": "Habitacion",
2023-01-31 13:35:25             "suma": {
2023-01-31 13:35:25                 "type": "command",
2023-01-31 13:35:25                 "value": [
2023-01-31 13:35:25                     3,
2023-01-31 13:35:25                     4
2023-01-31 13:35:25                 ],
2023-01-31 13:35:25                 "metadata": {}
2023-01-31 13:35:25             }
2023-01-31 13:35:25         }
2023-01-31 13:35:25     ],
2023-01-31 13:35:25     "actionType": "update"
2023-01-31 13:35:25 }
2023-01-31 13:35:25 
2023-01-31 13:35:25  | comp=IoTAgent
2023-01-31 13:35:25 time=2023-01-31T12:35:25.736Z | lvl=DEBUG | corr=bca84fd4-a163-11ed-bd94-0242ac140003; cbfwd=1 | trans=32ac83b6-1f8c-45d2-ba55-319dee4f7e16 | op=IoTAgentNGSI.ContextServer-v2 | from=n/a | srv=opcua_car | subsrv=/demo | msg=Handling v2 update from [iotagent-opcua:4041] | comp=IoTAgent
2023-01-31 13:35:25 time=2023-01-31T12:35:25.737Z | lvl=DEBUG | corr=bca84fd4-a163-11ed-bd94-0242ac140003; cbfwd=1 | trans=32ac83b6-1f8c-45d2-ba55-319dee4f7e16 | op=IoTAgentNGSI.ContextServer-v2 | from=n/a | srv=opcua_car | subsrv=/demo | msg={
2023-01-31 13:35:25     "entities": [
2023-01-31 13:35:25         {
2023-01-31 13:35:25             "id": "Habitacion:001",
2023-01-31 13:35:25             "type": "Habitacion",
2023-01-31 13:35:25             "suma": {
2023-01-31 13:35:25                 "type": "command",
2023-01-31 13:35:25                 "value": [
2023-01-31 13:35:25                     3,
2023-01-31 13:35:25                     4
2023-01-31 13:35:25                 ],
2023-01-31 13:35:25                 "metadata": {}
2023-01-31 13:35:25             }
2023-01-31 13:35:25         }
2023-01-31 13:35:25     ],
2023-01-31 13:35:25     "actionType": "update"
2023-01-31 13:35:25 } | comp=IoTAgent
2023-01-31 13:35:25 time=2023-01-31T12:35:25.737Z | lvl=DEBUG | corr=bca84fd4-a163-11ed-bd94-0242ac140003; cbfwd=1 | trans=32ac83b6-1f8c-45d2-ba55-319dee4f7e16 | op=IoTAgentNGSI.MongoDBDeviceRegister | from=n/a | srv=opcua_car | subsrv=/demo | msg=Looking for device with [{"name":"Habitacion:001","service":"opcua_car","subservice":"/demo","type":"Habitacion"}]. | comp=IoTAgent
2023-01-31 13:35:25 time=2023-01-31T12:35:25.738Z | lvl=DEBUG | corr=b53fa143-87a7-495e-8410-1fb319717ab5 | trans=b53fa143-87a7-495e-8410-1fb319717ab5 | op=IoTAgentNGSI.MongoDBGroupRegister | from=n/a | srv=n/a | subsrv=n/a | msg=Looking for group params ["service","subservice","type","apikey"] with queryObj {"service":"opcua_car","subservice":"/demo","type":"Habitacion","apikey":"iot"} | comp=IoTAgent
2023-01-31 13:35:25 time=2023-01-31T12:35:25.740Z | lvl=DEBUG | corr=b53fa143-87a7-495e-8410-1fb319717ab5 | trans=b53fa143-87a7-495e-8410-1fb319717ab5 | op=IoTAgentNGSI.MongoDBGroupRegister | from=n/a | srv=n/a | subsrv=n/a | msg=Device group for fields [["service","subservice","type","apikey"]] not found: [{"service":"opcua_car","subservice":"/demo","type":"Habitacion","apikey":"iot"}] | comp=IoTAgent
2023-01-31 13:35:25 time=2023-01-31T12:35:25.740Z | lvl=DEBUG | corr=b53fa143-87a7-495e-8410-1fb319717ab5 | trans=b53fa143-87a7-495e-8410-1fb319717ab5 | op=IoTAgentNGSI.MongoDBGroupRegister | from=n/a | srv=n/a | subsrv=n/a | msg=Looking for group params ["service","subservice","apikey"] with queryObj {"service":"opcua_car","subservice":"/demo","apikey":"iot"} | comp=IoTAgent
2023-01-31 13:35:25 time=2023-01-31T12:35:25.741Z | lvl=DEBUG | corr=b53fa143-87a7-495e-8410-1fb319717ab5 | trans=b53fa143-87a7-495e-8410-1fb319717ab5 | op=IoTAgentNGSI.MongoDBGroupRegister | from=n/a | srv=opcua_car | subsrv=/demo | msg=Device group data found: {"_id":"63d905c02068596abb6aeed0","resource":"/iot/opcua","apikey":"iot","type":"Device","service":"opcua_car","subservice":"/demo"} | comp=IoTAgent
2023-01-31 13:35:25 time=2023-01-31T12:35:25.741Z | lvl=DEBUG | corr=b53fa143-87a7-495e-8410-1fb319717ab5 | trans=b53fa143-87a7-495e-8410-1fb319717ab5 | op=IoTAgentNGSI.DeviceService | from=n/a | srv=n/a | subsrv=n/a | msg=deviceData before merge with conf: {"_id":"63d905c020685961af6aeed9","lazy":[],"active":[{"name":"temperature","type":"Float","object_id":"temperature"},{"name":"pressure","type":"Integer","object_id":"pressure"}],"commands":[{"name":"temperatureCommand","type":"command","object_id":"temperatureCommand"},{"name":"suma","type":"command","object_id":"suma"},{"name":"ajusteTemperatura","type":"command","ob [... cut ...]
2023-01-31 13:43:09 time=2023-01-31T12:43:09.778Z | lvl=DEBUG | corr=b53fa143-87a7-495e-8410-1fb319717ab5 | trans=b53fa143-87a7-495e-8410-1fb319717ab5 | op=IoTAgentOPCUA.CommonBindings | from=n/a | srv=n/a | subsrv=n/a | msg=Single measure for device [Habitacion:001] with apiKey [iot] successfully updated | comp=IoTAgent | service=opcua_car | subservice=/demo
2023-01-31 13:43:09 time=2023-01-31T12:43:09.779Z | lvl=DEBUG | corr=b53fa143-87a7-495e-8410-1fb319717ab5 | trans=b53fa143-87a7-495e-8410-1fb319717ab5 | op=IoTAgentNGSI.DomainControl | from=n/a | srv=n/a | subsrv=n/a | msg=response-time: 1244139 | comp=IoTAgent

```

manfredipist, 2023-02-02:
Hi @alberto072710
There's a little issue with the body you're sending. You're sending the following
```json
{
  type: 'command',
  value: [3,4]
}           
```
But you should send:
```json
{
  type: 'command',
  value: '[3,4]'
}           
```

alberto072710, 2023-02-02:
Thank you. You were right. In the status entity I get Ok but in the info entity I get null, I don't get the output parameter

manfredipist, 2023-02-02:
Does any error get printed in the logs? can you check please?

alberto072710, 2023-02-02:
This is the log I get when calling the method. A maybe stupid question, do we have to declare the outputArguments in the config.js file?

```
2023-02-02 13:46:52 time=2023-02-02T12:46:52.286Z | lvl=DEBUG | corr=aab268ec-a2f7-11ed-a683-0242ac180003; cbfwd=1 | trans=81ef9f20-e4b6-46d0-a645-f3441a13055e | op=IoTAgentNGSI.GenericMiddlewares | from=n/a | srv=opcua_car | subsrv=/demo | msg=Request for path [//op/update] from [iotagent-opcua:4041] | comp=IoTAgent
2023-02-02 13:46:52 time=2023-02-02T12:46:52.286Z | lvl=DEBUG | corr=aab268ec-a2f7-11ed-a683-0242ac180003; cbfwd=1 | trans=81ef9f20-e4b6-46d0-a645-f3441a13055e | op=IoTAgentNGSI.GenericMiddlewares | from=n/a | srv=opcua_car | subsrv=/demo | msg=Body:
2023-02-02 13:46:52 
2023-02-02 13:46:52 {
2023-02-02 13:46:52     "entities": [
2023-02-02 13:46:52         {
2023-02-02 13:46:52             "id": "Habitacion:001",
2023-02-02 13:46:52             "type": "Habitacion",
2023-02-02 13:46:52             "suma": {
2023-02-02 13:46:52                 "type": "command",
2023-02-02 13:46:52                 "value": "[3,4]",
2023-02-02 13:46:52                 "metadata": {}
2023-02-02 13:46:52             }
2023-02-02 13:46:52         }
2023-02-02 13:46:52     ],
2023-02-02 13:46:52     "actionType": "update"
2023-02-02 13:46:52 }
2023-02-02 13:46:52 
2023-02-02 13:46:52  | comp=IoTAgent
2023-02-02 13:46:52 time=2023-02-02T12:46:52.287Z | lvl=DEBUG | corr=aab268ec-a2f7-11ed-a683-0242ac180003; cbfwd=1 | trans=81ef9f20-e4b6-46d0-a645-f3441a13055e | op=IoTAgentNGSI.ContextServer-v2 | from=n/a | srv=opcua_car | subsrv=/demo | msg=Handling v2 update from [iotagent-opcua:4041] | comp=IoTAgent
2023-02-02 13:46:52 time=2023-02-02T12:46:52.288Z | lvl=DEBUG | corr=aab268ec-a2f7-11ed-a683-0242ac180003; cbfwd=1 | trans=81ef9f20-e4b6-46d0-a645-f3441a13055e | op=IoTAgentNGSI.ContextServer-v2 | from=n/a | srv=opcua_car | subsrv=/demo | msg={
2023-02-02 13:46:52     "entities": [
2023-02-02 13:46:52         {
2023-02-02 13:46:52             "id": "Habitacion:001",
2023-02-02 13:46:52             "type": "Habitacion",
2023-02-02 13:46:52             "suma": {
2023-02-02 13:46:52                 "type": "command",
2023-02-02 13:46:52                 "value": "[3,4]",
2023-02-02 13:46:52                 "metadata": {}
2023-02-02 13:46:52             }
2023-02-02 13:46:52         }
2023-02-02 13:46:52     ],
2023-02-02 13:46:52     "actionType": "update"
2023-02-02 13:46:52 } | comp=IoTAgent
2023-02-02 13:46:52 time=2023-02-02T12:46:52.288Z | lvl=DEBUG | corr=aab268ec-a2f7-11ed-a683-0242ac180003; cbfwd=1 | trans=81ef9f20-e4b6-46d0-a645-f3441a13055e | op=IoTAgentNGSI.MongoDBDeviceRegister | from=n/a | srv=opcua_car | subsrv=/demo | msg=Looking for device with [{"name":"Habitacion:001","service":"opcua_car","subservice":"/demo","type":"Habitacion"}]. | comp=IoTAgent
2023-02-02 13:46:52 time=2023-02-02T12:46:52.292Z | lvl=DEBUG | corr=46fb53ba-d8f7-4ad0-b7bf-dca196f58193 | trans=46fb53ba-d8f7-4ad0-b7bf-dca196f58193 | op=IoTAgentNGSI.MongoDBGroupRegister | from=n/a | srv=n/a | subsrv=n/a | msg=Looking for group params ["service","subservice","type","apikey"] with queryObj {"service":"opcua_car","subservice":"/demo","type":"Habitacion","apikey":"iot"} | comp=IoTAgent
2023-02-02 13:46:52 time=2023-02-02T12:46:52.296Z | lvl=DEBUG | corr=46fb53ba-d8f7-4ad0-b7bf-dca196f58193 | trans=46fb53ba-d8f7-4ad0-b7bf-dca196f58193 | op=IoTAgentNGSI.MongoDBGroupRegister | from=n/a | srv=n/a | subsrv=n/a | msg=Device group for fields [["service","subservice","type","apikey"]] not found: [{"service":"opcua_car","subservice":"/demo","type":"Habitacion","apikey":"iot"}] | comp=IoTAgent
2023-02-02 13:46:52 time=2023-02-02T12:46:52.296Z | lvl=DEBUG | corr=46fb53ba-d8f7-4ad0-b7bf-dca196f58193 | trans=46fb53ba-d8f7-4ad0-b7bf-dca196f58193 | op=IoTAgentNGSI.MongoDBGroupRegister | from=n/a | srv=n/a | subsrv=n/a | msg=Looking for group params ["service","subservice","apikey"] with queryObj {"service":"opcua_car","subservice":"/demo","apikey":"iot"} | comp=IoTAgent
2023-02-02 13:46:52 time=2023-02-02T12:46:52.304Z | lvl=DEBUG | corr=46fb53ba-d8f7-4ad0-b7bf-dca196f58193 | trans=46fb53ba-d8f7-4ad0-b7bf-dca196f58193 | op=IoTAgentNGSI.MongoDBGroupRegister | from=n/a | srv=opcua_car | subsrv=/demo | msg=Device group data found: {"_id":"63dbb096d0f25178bbb301d1","resource":"/iot/opcua","apikey":"iot","type":"Device","service":"opcua_car","subservice":"/demo"} | comp=IoTAgent
2023-02-02 13:46:52 time=2023-02-02T12:46:52.305Z | lvl=DEBUG | corr=46fb53ba-d8f7-4ad0-b7bf-dca196f58193 | trans=46fb53ba-d8f7-4ad0-b7bf-dca196f58193 | op=IoTAgentNGSI.DeviceService | from=n/a | srv=n/a | subsrv=n/a | msg=deviceData before merge with conf: {"_id":"63dbb096d0f2517857b301da","lazy":[],"active":[{"name":"temperature","type":"Float","object_id":"temperature"},{"name":"pressure","type":"Integer","object_id":"pressure"}],"commands":[{"name":"temperatureCommand","type":"command","object_id":"temperatureCommand"},{"name":"suma","type":"command","object_id":"suma"},{"name":"ajusteTemperatura","type":"command","object_id":"ajusteTemperatura"}],"staticAttributes":[],"subscriptions":[],"creationDate":"2023-02-02T12:46:14.523Z","id":"Habitacion:001","type":"Habitacion","name":"Habitacion:001","service":"opcua_car","subservice":"/demo","registrationId":"63dbb09683d292467f24fd3b","internalId":null,"apikey":"iot","endpoint":"opc.tcp://192.168.2.1:4840","explicitAttrs":false} defaults: [null,null,[],[],[],[],[]] fields: ["lazy","internalAttr [... cut ...]
2023-02-02 13:46:52 time=2023-02-02T12:46:52.428Z | lvl=INFO | corr=46fb53ba-d8f7-4ad0-b7bf-dca196f58193 | trans=46fb53ba-d8f7-4ad0-b7bf-dca196f58193 | op=IoTAgentOPCUA.OPCUABinding | from=n/a | srv=n/a | subsrv=n/a | msg=Update to execute =[{"nodeId":"ns=4;i=9","attributeId":13,"indexRange":null,"value":{"value":{"dataType":0,"value":"[3,4]"}}}] | comp=IoTAgent
2023-02-02 13:46:56 time=2023-02-02T12:46:56.004Z | lvl=INFO | corr=46fb53ba-d8f7-4ad0-b7bf-dca196f58193 | trans=46fb53ba-d8f7-4ad0-b7bf-dca196f58193 | op=IoTAgentOPCUA.OPCUABinding | from=n/a | srv=n/a | subsrv=n/a | msg=subscription keepalive | comp=IoTAgent

```

walterwootz, 2023-02-22:
Hi @alberto072710, sorry for the delay. Maybe the issue is related to OPCUA server implementation. Can you provide us your test server in order to find the issue?

alberto072710, 2023-02-23:
Hi,
The delay in answering does not matter. I was currently using a S7-1200 Siemens PLC, to test that it was not a problem with the PLC server I am trying to test the PROSYS OPC server which provides an example method with 2 input and 1 output parameters. I'm trying to send the example command but I get an error because the first input parameter is of type string (I don't know if this has something to do with it). I am attaching the data so you can test it.

docker_compose.yml

```
version: "3.1"

volumes:
  mongodb: ~

services:
  iot-agent:
    image: iotagent4fiware/iotagent-opcua:2.1.2
    hostname: iotagent-opcua
    depends_on:
      - mongodb
      - orion
    networks:
      - hostnet
    expose:
      - "4041"
      - "9229"
    ports:
      - "4041:4041"
      - "9229:9229"
    environment:
      - "IOTA_CB_HOST=orion"
      - "IOTA_CB_PORT=1026"
      - "IOTA_NORTH_PORT=4041"
      - "IOTA_REGISTRY_TYPE=mongodb"
      - "IOTA_MONGO_HOST=mongodb"
      - "IOTA_MONGO_PORT=27017"
      - "IOTA_MONGO_DB=iotagent_opcua"
      - "IOTA_PROVIDER_URL=http://iotagent-opcua:4041"
      - "IOTA_OPCUA_ENDPOINT=opc.tcp://DESKTOP-9P47CVL:53530/OPCUA/SimulationServer"
      - "IOTA_OPCUA_SECURITY_MODE=None"
      - "IOTA_OPCUA_SECURITY_POLICY=None"
      #- "IOTA_OPCUA_SECURITY_USERNAME="
      #- "IOTA_OPCUA_SECURITY_PASSWORD="
      - "IOTA_OPCUA_UNIQUE_SUBSCRIPTION=false"
      - "IOTA_OPCUA_SUBSCRIPTION_NOTIFICATIONS_PER_PUBLISH=1000"
      - "IOTA_OPCUA_SUBSCRIPTION_PUBLISHING_ENABLED=true"
      - "IOTA_OPCUA_SUBSCRIPTION_REQ_LIFETIME_COUNT=100"
      - "IOTA_OPCUA_SUBSCRIPTION_REQ_MAX_KEEP_ALIVE_COUNT=10"
      - "IOTA_OPCUA_SUBSCRIPTION_REQ_PUBLISHING_INTERVAL=1000"
      - "IOTA_OPCUA_SUBSCRIPTION_PRIORITY=128"
      - "IOTA_EXTENDED_FORBIDDEN_CHARACTERS=[]"
      - "IOTA_OPCUA_MT_POLLING=false"
      - "IOTA_OPCUA_MT_AGENT_ID=age01_"
      - "IOTA_OPCUA_MT_ENTITY_ID=age01_Car"
      - "IOTA_OPCUA_MT_ENTITY_TYPE=Habitacion"
      - "IOTA_OPCUA_MT_NAMESPACE_IGNORE=0,7"
      - "IOTA_OPCUA_MT_STORE_OUTPUT=true"
    volumes:
      - C:/Users/Alberto/Desktop/OPC_UA_SIN_CERT/prosys/conf/config.js:/opt/iotagent-opcua/conf/config.js

      
  mongodb:
    image: mongo:4.2
    hostname: mongodb
    networks:
      - hostnet
    ports:
      - "27017:27017"
    command: --bind_ip_all
    volumes:
      - mongodb:/data
  ################ OCB ################

[... 275 lines cut ...]
2023-02-23 11:07:18 time=2023-02-23T10:07:18.716Z | lvl=ERROR | corr=a401585d-113a-4bbf-9109-7d5f76f833ab | trans=a401585d-113a-4bbf-9109-7d5f76f833ab | op=IoTAgentOPCUA.CommandHandler | from=n/a | srv=opcua_car | subsrv=/demo | msg=COMMAND-003: Command execution could not be handled, [TypeError: Cannot read properties of null (reading '0')
2023-02-23 11:07:18     at AsyncFunction.executeCommand (/opt/iotagent-opcua/lib/bindings/OPCUABinding.js:386:53)
2023-02-23 11:07:18     at runMicrotasks (<anonymous>)
2023-02-23 11:07:18     at processTicksAndRejections (node:internal/process/task_queues:96:5)] | comp=IoTAgent
2023-02-23 11:07:18 time=2023-02-23T10:07:18.718Z | lvl=INFO | corr=a401585d-113a-4bbf-9109-7d5f76f833ab | trans=a401585d-113a-4bbf-9109-7d5f76f833ab | op=IoTAgentOPCUA.OPCUABinding | from=n/a | srv=opcua_car | subsrv=/demo | msg=ns=6;s=MyMethod read variable type -> { /* DataValue */
2023-02-23 11:07:18    value: Variant(Scalar<Null>, value: <null>)
2023-02-23 11:07:18    statusCode:      BadAttributeIdInvalid (0x80350000)
2023-02-23 11:07:18    serverTimestamp: null
2023-02-23 11:07:18    sourceTimestamp: null
2023-02-23 11:07:18 } | comp=IoTAgent
2023-02-23 11:07:18 time=2023-02-23T10:07:18.718Z | lvl=INFO | corr=a401585d-113a-4bbf-9109-7d5f76f833ab | trans=a401585d-113a-4bbf-9109-7d5f76f833ab | op=IoTAgentOPCUA.OPCUABinding | from=n/a | srv=opcua_car | subsrv=/demo | msg=Update to execute =[{"nodeId":"ns=6;s=MyMethod","attributeId":13,"indexRange":null,"value":{"value":{"dataType":0,"value":"[sin,7]"}}}] | comp=IoTAgent
2023-02-23 11:07:18 time=2023-02-23T10:07:18.719Z | lvl=DEBUG | corr=a401585d-113a-4bbf-9109-7d5f76f833ab | trans=a401585d-113a-4bbf-9109-7d5f76f833ab | op=IoTAgentNGSI.Request | from=n/a | srv=opcua_car | subsrv=/demo | msg=Response "" | comp=IoTAgent
2023-02-23 11:07:18 time=2023-02-23T10:07:18.719Z | lvl=INFO | corr=a401585d-113a-4bbf-9109-7d5f76f833ab | trans=a401585d-113a-4bbf-9109-7d5f76f833ab | op=IoTAgentNGSI.Entities-v2 | from=n/a | srv=opcua_car | subsrv=/demo | msg=Received the following response from the CB: Value updated successfully
2023-02-23 11:07:18  | comp=IoTAgent
2023-02-23 11:07:18 time=2023-02-23T10:07:18.719Z | lvl=DEBUG | corr=a401585d-113a-4bbf-9109-7d5f76f833ab | trans=a401585d-113a-4bbf-9109-7d5f76f833ab | op=IoTAgentNGSI.ContextServer-v2 | from=n/a | srv=opcua_car | subsrv=/demo | msg=Update action from [iotagent-opcua:4041] handled successfully. | comp=IoTAgent
2023-02-23 11:07:18 time=2023-02-23T10:07:18.720Z | lvl=DEBUG | corr=db12afc2-b361-11ed-ad7e-0242ac150003; cbfwd=1 | trans=4a789694-b997-478e-a282-dbeb0d5a5380 | op=IoTAgentNGSI.DomainControl | from=n/a | srv=opcua_car | subsrv=/demo | msg=response-time: 20 | comp=IoTAgent
2023-02-23 11:07:22 time=2023-02-23T10:07:22.027Z | lvl=INFO | corr=a401585d-113a-4bbf-9109-7d5f76f833ab | trans=a401585d-113a-4bbf-9109-7d5f76f833ab | op=IoTAgentOPCUA.OPCUABinding | from=n/a | srv=opcua_car | subsrv=/demo | msg=subscription keepalive | comp=IoTAgent
```

PROSYS OPC UA SIMULATION SERVER

[image]
[image]

Thanks!

manfredipist, 2023-02-28:
Hi @alberto072710 , fixed with in v2.1.3. One minor change is that now input arguments MUSTbe passed as an array, eg:
```json
{
  type: 'command',
  value: [3,4]
}    
```

alberto072710, 2023-03-01:
Hi, 
Amazing work. Now it works perfectly. Thanks a lot

closed 2023-03-01 by alberto072710