Engineering-Research-and-Development/iotagent-opcua #98: Disabling Mapping Tool

drypatrick, 2023-01-31:
I would like to know if there is a way to disable the mapping tool, since I would like to provisioning the devices and their service groups, programmatically.

Thanks.

drypatrick, 2023-02-02:
I found out how to disable the mapping tool, placing null o empty array as value in the following field, of the config.js:
```
    /**
     * Types array for static configuration of services. Check documentation in the IoT Agent Library for Node.js for
     *  further details:
     *
     *      https://github.com/Engineering-Research-and-Development/iotagent-opcua#type-configuration
     */
    types: {
		null: {
			active: [
				{
					name: null,
					type: null
				}
			],
			lazy: [],
			commands: []
		}
	},
	contexts: [],
	contextSubscriptions: [],
```
But which is the correct syntax to provision a device trough API?  I can't figure it out how to give information about the contex, and I'm using this call which create the entity but of course is not working:

```
curl -iX POST \
'http://localhost:4041/iot/devices' \
-H 'Content-Type: application/json' \
-H 'fiware-service: opcua_car' \
-H 'fiware-servicepath: /demo' \
-d '{
	"devices": [
		{
			"device_id": "urn:ngsiv2:I40Asset:PLC:001",
			"type": "PLC",
			"attributes": [
				{
					"name": "processStatus",
					"type": "Text"
				}
			]
		}
	]
}
'
```

walterwootz, 2023-02-02:
Hi @drypatrick, you can disable the mapping by setting `config.autoprovision = false;`
I also recommend you to set also empty arrays for the followings:
```js
types: [],
contexts: [],
contextSubscriptions: []
```
About  provision a device trough API, you should perform the service group provisioning and then the device provisioning.
You can see requests body respectively on [performGroupProvisioning.request.json](https://github.com/Engineering-Research-and-Development/iotagent-opcua/blob/master/test/functional/mock/performGroupProvisioning.request.json) and [performDeviceProvisioning.request.json](https://github.com/Engineering-Research-and-Development/iotagent-opcua/blob/master/test/functional/mock/performDeviceProvisioning.request.json).

You can have a look on functional test [here](https://github.com/Engineering-Research-and-Development/iotagent-opcua/blob/master/test/functional/apiProvisioning.test.js) (lines 77 and 95) to see how to perform the requests.

drypatrick, 2023-02-02:
I'm provisioning the service group, with this call:
```
curl -iX POST \
'http://localhost:4041/iot/services' \
-H 'Content-Type: application/json' \
-H 'fiware-service: opcua_car' \
-H 'fiware-servicepath: /demo' \
-d '{ "services": [{ "apikey": "4jggokgpepnvsb2uv4s40d59ov", "cbroker": "http://orion:1026", "entity_type": "PLC", "resource": "/iot/d" }] }'
```
And the device with this one:
```
curl -iX POST \
'http://localhost:4041/iot/devices' \
-H 'Content-Type: application/json' \
-H 'fiware-service: opcua_car' \
-H 'fiware-servicepath: /demo' \
-d '{
"devices": [
    {
      "entity_name": "urn:ngsiv2:I40Asset:PLC:001",
      "entity_type": "PLC",
      "device_id": "urn:ngsiv2:I40Asset:PLC:001",
      "apikey": "urn:ngsiv2:I40Asset:PLC:001",
      "endpoint": "opc.tcp://10.0.7.236:4840/",
      "attributes": [
        {
          "object_id": "processStatus",
          "name": "processStatus",
          "type": "Text"
        }
      ],
      "lazy": [],
      "commands": [],
      "static_attributes": [],
      "explicitAttrs": false
    }
  ]
}'
```
But I should also supply the `opcua_id` somewhere, isn't it? As it's done in the `config.js`
```

...
contexts: [
	{
		id: 'urn:ngsiv2:I40Asset:PLC:001',
		type: 'PLC',
		mappings: [
			{
				ocb_id: 'processStatus',
				opcua_id: 'ns=4;i=198',
				inputArguments: []
			}
		]
	}
...
],
```

walterwootz, 2023-02-02:
Southbound provisioning through API is not implemented yet since the IoTAgent-OPCUA can be connected only to one OPCUA Server. I suggest you going through a manual provisioning using static config.js and handle the northbound provisioning as you prefer.

We will consider further the Southbound provisioning API in future developments.

drypatrick, 2023-02-02:
So is not possible to provision the agent trough API, is that correct?

walterwootz, 2023-02-22:
Such API is not planned to be implemented, since the agent is able to connect only to one device per time. Please go through manual southbound provisioning using config.js. In a multi-device scenario, you could run up different instances of the agent.

closed 2023-02-22 by walterwootz