Engineering-Research-and-Development/iotagent-opcua #110: Request entity too large

IngMiad, 2023-04-19:
Hello,

we started using the opc ua iot agent to get/write data from opc ua datasources. It's working fine for smaller datasources, but our newest datasource brought up some issues because of its size.

The newest datasource is a process control system which has about 4000 relevant Tags, that we want to use (from about 50000 available tags overall). Because creating a manual configuration for the huge amount of tags is a lot of work, we created a tool that a) filters the tags b) groups them into logical entities based on name schema c) creates a FIWARE smart data model and d) automatically generates a FIWARE OPC UA IoT-Agent configuration to read the tags. The result is an agent configuration that covers about 55 entities with about 4000 active attributes.

Testing the result we ran into a "Request Entity too large" issue (see attached image). **Could you give us some advice what this error is exactly about and where are the limits of the request entity size by OPC UA IoT-Agent?** 
According to limitation our plan is to rewrite our tool, so that the 4000 relevant tags will be queried by multiple OPC UA IoT-Agents. I already tested the generated agent configuration with only 3 entities which seem to work fine. 

[image]

Thank you in advance!
Greetings, Inga

manfredipist, 2023-04-19:
Hello @IngMiad ,
It seems that this issue you're encountering is a known limitation of Orion Context Broker [Link to Doc](https://github.com/telefonicaid/fiware-orion/blob/0dc77185b8f98274840abc28364e2bd071c1d7e4/doc/manuals/admin/perf_tuning.md#payload-and-message-size-and-performance)
Luckly you can overwrite this behaviour by overwriting _inReqPayloadMaxSize_ paramter in your docker compose file as follows:
```yaml
....
 command: -dbhost mongo-db -inReqPayloadMaxSize {BYTE} -logLevel DEBUG 
...
```
In this case you should replace BYTE with the proper value according to your use case and performances degradation

Let us know if this solves your issue
Best Regards!

IngMiad, 2023-04-21:
Hello @manfredipist,

thank you very much for the quick reply. I am glad it is a configurable limitation of Orion Context Broker.

I will test our scenario again with the changes in the next 2-3 weeks (as soon as I will visit our plant again) and will inform you about the success.

Greetings!

manfredipist, 2023-07-27:
Closing as stale.

Please consider spending few minutes to fulfill the following form to help us improving the IoT Agent, thanks!
[Form link](https://forms.office.com/e/aM6hS3xmE4)

IngMiad, 2023-09-28:
Hello as an Update: This feature is only available in Orion NGSIv2 from Telefonica (https://github.com/telefonicaid/fiware-orion) not in Orion-LD. 

The feature has also been ported to Orion-LD now, but doesn't seem to work as expected and isn't solved until now. As a reference you can check following issue: https://github.com/FIWARE/context.Orion-LD/issues/1356

IngMiad, 2024-07-23:
Hello,

long time ago, but I have a few updates on this issue. I have been talking to Orion-LD developer who added the feature to adjust -inReqPayloadMaxSize also in Orion-LD. Although I am configuring the -inReqPayloadMaxSize and set it to a huge value (1TB), I am still getting the error. 

After a few further tests I doubt that the request size limitation is coming from Orion-LD. 
1. I configured debug mode for IoT-Agent OPC UA and Orion-LD
2. Afterwards I started OPC UA Agent with my configuration
3. In the IoT-Agent logs I can see the issue that the request is too large   
    [image]
5. At Orion-LD side I don't see any incoming request in the logs at all (besides Kubernetes Health-Check at version endpoint)

Here are the full IoT-Agent Logs:
```
{"time":"2024-07-18T13:49:18.504Z","lvl":"ERROR","msg":"EISDIR: illegal operation on a directory, read"}
{"op":"IoTAgentNGSI.JEXL","time":"2024-07-18T13:49:18.804Z","lvl":"INFO","msg":"Trasformations can be added to JEXL parser"}
{"time":"2024-07-18T13:49:19.350Z","lvl":"INFO","msg":"Setting IOTA_CB_NGSI_VERSION to environment value: ld"}
{"time":"2024-07-18T13:49:19.350Z","lvl":"INFO","msg":"Setting IOTA_REGISTRY_TYPE to environment value: mongodb"}
{"time":"2024-07-18T13:49:19.350Z","lvl":"INFO","msg":"Setting IOTA_LOG_LEVEL to environment value: DEBUG"}
{"time":"2024-07-18T13:49:19.350Z","lvl":"INFO","msg":"Setting IOTA_MONGO_HOST to environment value: iot-mongo-db"}
{"time":"2024-07-18T13:49:19.350Z","lvl":"INFO","msg":"Setting IOTA_MONGO_DB to environment value: iotagent_opcua"}
{"time":"2024-07-18T13:49:19.350Z","lvl":"INFO","msg":"Setting IOTA_RELAX_TEMPLATE_VALIDATION to environment value: true"}
{"time":"2024-07-18T13:49:19.350Z","lvl":"INFO","msg":"Setting IOTA_JSON_LD_CONTEXT to environment value: http://ld-context/ld-context/datamodels.context-ngsild.jsonld"}
{"time":"2024-07-18T13:49:19.350Z","lvl":"INFO","msg":"Setting IOTA_FALLBACK_TENANT to environment value: opcua_pcs"}
{"time":"2024-07-18T13:49:19.350Z","lvl":"WARN","msg":"***********************************************\nWARNING: authentication for secure connections is not in use,\nIt is recommended to enable authentication\n***********************************************"}
time=2024-07-18T13:49:19.350Z | lvl=INFO | corr=5458d490-3eac-4c63-a2e0-4d977a56f9e7 | trans=5458d490-3eac-4c63-a2e0-4d977a56f9e7 | op=IoTAgentNGSI.Global | from=n/a | srv=n/a | subsrv=n/a | msg=Starting an IoT Agent... | comp=IoTAgent
time=2024-07-18T13:49:19.351Z | lvl=INFO | corr=5458d490-3eac-4c63-a2e0-4d977a56f9e7 | trans=5458d490-3eac-4c63-a2e0-4d977a56f9e7 | op=IoTAgentNGSI.Global | from=n/a | srv=n/a | subsrv=n/a | msg=Setting IOTA_CB_NGSI_VERSION to environment value: ld | comp=IoTAgent
time=2024-07-18T13:49:19.352Z | lvl=INFO | corr=5458d490-3eac-4c63-a2e0-4d977a56f9e7 | trans=5458d490-3eac-4c63-a2e0-4d977a56f9e7 | op=IoTAgentNGSI.Global | from=n/a | srv=n/a | subsrv=n/a | msg=Setting IOTA_REGISTRY_TYPE to environment value: mongodb | comp=IoTAgent
time=2024-07-18T13:49:19.352Z | lvl=INFO | corr=5458d490-3eac-4c63-a2e0-4d977a56f9e7 | trans=5458d490-3eac-4c63-a2e0-4d977a56f9e7 | op=IoTAgentNGSI.Global | from=n/a | srv=n/a | subsrv=n/a | msg=Setting IOTA_LOG_LEVEL to environment value: DEBUG | comp=IoTAgent
time=2024-07-18T13:49:19.352Z | lvl=INFO | corr=5458d490-3eac-4c63-a2e0-4d977a56f9e7 | trans=5458d490-3eac-4c63-a2e0-4d977a56f9e7 | op=IoTAgentNGSI.Global | from=n/a | srv=n/a | subsrv=n/a | msg=Setting IOTA_MONGO_HOST to environment value: iot-mongo-db | comp=IoTAgent
time=2024-07-18T13:49:19.352Z | lvl=INFO | corr=5458d490-3eac-4c63-a2e0-4d977a56f9e7 | trans=5458d490-3eac-4c63-a2e0-4d977a56f9e7 | op=IoTAgentNGSI.Global | from=n/a | srv=n/a | subsrv=n/a | msg=Setting IOTA_MONGO_DB to environment value: iotagent_opcua | comp=IoTAgent
time=2024-07-18T13:49:19.352Z | lvl=INFO | corr=5458d490-3eac-4c63-a2e0-4d977a56f9e7 | trans=5458d490-3eac-4c63-a2e0-4d977a56f9e7 | op=IoTAgentNGSI.Global | from=n/a | srv=n/a | subsrv=n/a | msg=Setting IOTA_RELAX_TEMPLATE_VALIDATION to environment value: true | comp=IoTAgent
time=2024-07-18T13:49:19.352Z | lvl=INFO | corr=5458d490-3eac-4c63-a2e0-4d977a56f9e7 | trans=5458d490-3eac-4c63-a2e0-4d977a56f9e7 | op=IoTAgentNGSI.Global | from=n/a | srv=n/a | subsrv=n/a | msg=Setting IOTA_JSON_LD_CONTEXT to environment value: http://ld-context/ld-context/datamodels.context-ngsild.jsonld | comp=IoTAgent
time=2024-07-18T13:49:19.352Z | lvl=INFO | corr=5458d490-3eac-4c63-a2e0-4d977a56f9e7 | trans=5458d490-3eac-4c63-a2e0-4d977a56f9e7 | op=IoTAgentNGSI.Global | from=n/a | srv=n/a | subsrv=n/a | msg=Setting IOTA_FALLBACK_TENANT to environment value: opcua_pcs | comp=IoTAgent
time=2024-07-18T13:49:19.352Z | lvl=WARN | corr=5458d490-3eac-4c63-a2e0-4d977a56f9e7 | trans=5458d490-3eac-4c63-a2e0-4d977a56f9e7 | op=IoTAgentNGSI.Global | from=n/a | srv=n/a | subsrv=n/a | msg=***********************************************
WARNING: authentication for secure connections is not in use,
It is recommended to enable authentication
*********************************************** | comp=IoTAgent
time=2024-07-18T13:49:19.352Z | lvl=INFO | corr=5458d490-3eac-4c63-a2e0-4d977a56f9e7 | trans=5458d490-3eac-4c63-a2e0-4d977a56f9e7 | op=IoTAgentNGSI.Global | from=n/a | srv=n/a | subsrv=n/a | msg=Default expression language not set, or invalid, using legacy configuration | comp=IoTAgent
[... cut ...]
time=2024-07-18T13:54:55.995Z | lvl=WARN | corr=508aea2e-450d-11ef-a8c6-7cc255269eb8 | trans=1721310386-176-00000000031 | from=192.168.42.3 | srv=pending | subsrv=<default> | comp=Orion | op=rest.cpp[1514]:connectionTreat | msg=orionldState.in.contentType: None
time=2024-07-18T13:54:55.995Z | lvl=WARN | corr=508aebfa-450d-11ef-a49b-7cc255269eb8 | trans=1721310386-176-00000000032 | from=192.168.42.3 | srv=pending | subsrv=<default> | comp=Orion | op=rest.cpp[1514]:connectionTreat | msg=orionldState.in.contentType: None
```

Are you sure there is no size limitation at the side of IoT-Agent OPC UA? Thank you in advance!

walterwootz, 2024-07-24:
Hi! First of all thank you for your investigation.
The size limit is related to the iotagent-node-lib on top of which this OPCUA Agent is built on.

The problem has been fixed with the latest IotAgent-OPCUA release (v2.2.7) that includes the latest iotagent-node-lib (4.5.0).
You can use the env variable **IOTA_EXPRESS_LIMIT** (if using docker) or **expressLimit** (if you are using custom config.js) as described [here](https://github.com/telefonicaid/iotagent-node-lib/blob/master/doc/admin.md#expresslimit).

The default value is set to 1M (1 Megabyte), just change it as you prefer.

IngMiad, 2024-07-25:
Hello,

thank you very much. That are good news for us. I will test it and let you know how it worked, once I am back at our site for testing. 

Greetings

IngMiad, 2024-07-26:
The latest release at Docker Hub is currently 2.2.6: https://registry.hub.docker.com/r/fiware/iotagent-opcua/tags?page=&page_size=&ordering=&name=

When do you plan to upload 2.2.7 to Docker Hub?

walterwootz, 2024-07-26:
The official fiware repository is not updated yet. As workaround you can use [this docker hub repository](https://hub.docker.com/r/iotagent4fiware/iotagent-opcua/tags) (you ca find the 2.2.7 also).

IngMiad, 2024-07-30:
Hello,

I tested the new release yesterday an can now successfully consume a large amount of entities and parameter (~50 entities, ~4000 parameters) with IoT-Agent OPC UA and provide them to Orion-LD. Orion-LD also successfully receives and persists them with no issue. Thank you very much!

Greetings
Inga

closed 2024-07-30 by IngMiad