Engineering-Research-and-Development/iotagent-opcua #111: OPC UA server useridentity

barry-fip, 2023-06-20:
Currently the V2 version of the agent is only supporting ANONYMOUS identities. Would it be possible to implement opcua.UserTokenType.UserName for the opc ua binding?

barry-fip, 2023-06-20:
Hi @manfredipist , I use the docker-compose file to set the environment variables that configure the opc ua connection. These are the main values.

      - IOTA_OPCUA_ENDPOINT=opc.tcp://{address}:4840
      - IOTA_OPCUA_SECURITY_MODE=SignAndEncrypt
      - IOTA_OPCUA_SECURITY_POLICY=Basic256Sha256
      - IOTA_OPCUA_SECURITY_USERNAME=admin
      - IOTA_OPCUA_SECURITY_PASSWORD={pwd}
      - 
I noticed that the config values for username and password are not used in the code to create the session.

manfredipist, 2023-07-04:
hi @barry-fip , it should be fixed on v2.1.7, please let us know if everything is now working.

manfredipist, 2023-07-18:
Closing as stale, if you need further assistance feel free to open a new issue.

closed 2023-07-18 by manfredipist