Engineering-Research-and-Development/iotagent-opcua #117: Command execution, device not found

fdrobnic, 2023-09-26:
I'm testing the command execution on the example docker setup and the NGSI-LD mode. The device is autoprovisioned and can be listed both on the Agent side and the Orion side. However, when I try to send a command to Orion, the error 500 is returned and the following appears in the log:

    iot-agent-opcua-iotagent-opcua-1  | time=2023-09-26T13:32:27.149Z | lvl=DEBUG | corr=n/a | trans=n/a | op=IoTAgentNGSI.ContextServer-LD | from=n/a | srv=n/a | subsrv=n/a | msg=There was an error handling the update action: DeviceNotFound {
    iot-agent-opcua-iotagent-opcua-1  |   name: 'DEVICE_NOT_FOUND',
    iot-agent-opcua-iotagent-opcua-1  |   message: 'No device was found with id:urn:ngsi-ld:Device:age01_Car',
    iot-agent-opcua-iotagent-opcua-1  |   code: 404
    iot-agent-opcua-iotagent-opcua-1  | }. | comp=IoTAgent

The ids in the databases are indeed different. In the Agent database, it is "age01_Car" and in the Orion database, it is "urn:ngsi-ld:Device:age01_Car".

manfredipist, 2023-09-27:
Hi @fdrobnic 
Could you please provide the config.js  and docker-compose.yaml files you use to run the test?
Thanks

fdrobnic, 2023-09-27:
Hi, both files attached.
[conf.zip](https://github.com/Engineering-Research-and-Development/iotagent-opcua/files/12741592/conf.zip)

walterwootz, 2023-09-28:
Hello @fdrobnic , could you also provide docker-compose or docker run command you are using to run Orion LD?

fdrobnic, 2023-09-28:
Hi, here it is, together with the corresponding env file. Thank you.

[conf_orion.zip](https://github.com/Engineering-Research-and-Development/iotagent-opcua/files/12747308/conf_orion.zip)

walterwootz, 2023-09-28:
Hello @fdrobnic unfortunately this feature is not supported at the moment due to a missing feature on Orion LD (query forwarding). As you can see [here](https://github.com/FIWARE/context.Orion-LD/issues/280) this feature is planned for version 3.0.0 of Orion LD. We have already pushed this issue to the maintainer of Orion LD.

closed 2023-09-28 by walterwootz

fdrobnic, 2023-09-28:
Thank you very much for your effort. We hope that the author will find some time to implement it as soon as possible.