Engineering-Research-and-Development/iotagent-opcua #120: Actuators

fdrobnic, 2023-10-13:
Reagrding the problem discussed in the issue [#117](https://github.com/Engineering-Research-and-Development/iotagent-opcua/issues/117), and the fact that the [IoT Agent for Ultralight](https://github.com/telefonicaid/iotagent-ul) does provide a working example of actuator control, would it be possible to implement a similar solution as a workaround that would make some demonstrations possible, at least? The solution might have limitations (be less scalable, ...), but anyway. It would help us a lot.

walterwootz, 2023-10-26:
Hello @fdrobnic 
After some tests. you should be able to run actuation (commands) with the following configuration:
- Orion-LD v1.4.0
- OPC UA Agent v2.2.0 (pushed today to dockerhub)

This is an example curl to run a command through Orion LD:

```sh
curl --location --request PATCH 'http://localhost:1026/ngsi-ld/v1/entities/urn:ngsi-ld:Device:age01_Car/attrs/Accelerate' \
--header 'fiware-service: opcua_car' \
--header 'Content-Type: application/json' \
--data-raw '{
    "value": "20"
}'
```

Everything should work correctly.

fdrobnic, 2023-11-03:
Thank you very much. It works as expected.

closed 2023-11-03 by fdrobnic