Engineering-Research-and-Development/iotagent-opcua #135: Compound vs. Numerical Value

IngMiad, 2024-09-27:
Hi there,

I have a questions regarding the data format of the payload that is send from OPC UA IoT-Agent to the nothbound interface (to Orion-LD in my case). 

#### Issue
In my scenario I am using two OPC UA IoT-Agents for 1) a weather station and 2) process control system. The payload send to the northbound interface from both IoT-Agents is different, so that data for the first data source is stored as numeric values and the data for the second data source is stored as compound values for short-time-history at Orion-LD. Orion-LD just stores the data as it comes in, so I compared the payloads from IoT-Agents and noticed that they are different (see https://github.com/FIWARE/context.Orion-LD/issues/1651). I'd like to store the data of both data sources as numeric values, so I am asking myself why the IoT-Agent sends data as composed values in the second case and not in the same format than the first data source. 

#### Payload from OPC UA Agent
Comparing the payload send from OPC UA agent to Orion-LD we can see that the data is provided as numeric value for the first data source and as compound value for the second data source from OPC UA agent. I got the payload from OPC UA agent logs…

Data send to Orion-LD for the first weather station source:
```
...
trans=5fe32dba-7267-4ff4-8276-31d875e3c7a7 | op=IoTAgentNGSI.Request | from=n/a | srv=n/a | subsrv=n/a | msg=Options: {
    "url": "http://orion-ld:1026/ngsi-ld/v1/entityOperations/upsert/?options=update",
    "method": "POST",
    "headers": {
        "fiware-service": "opcua_meteo",
        "fiware-servicepath": "/weather",
        "Content-Type": "application/ld+json",
        "NGSILD-Tenant": "opcua_meteo",
        "NGSILD-Path": "/weather"
    },
    "json": [
        {
            "@context": "http://ld-context/ld-context/datamodels.context-ngsild.jsonld",
            "temperature": {
                "type": "Property",
                "value": 46252,
                "observedAt": "2024-09-26T15:46:16.925Z"
            },
            "id": "urn:ngsi-ld:WeatherObserved:meteostation-01",
            "type": "WeatherObserved"
        }
    ]
} | comp=IoTAgent
...

```

The data from the second process control station data source is provided to Orion-LD like that:
```
...
trans=89b57f04-8b90-42f1-bd47-11fdd947af41 | op=IoTAgentNGSI.Request | from=n/a | srv=n/a | subsrv=n/a | msg=Options: {
    "url": "http://orion-ld:1026/ngsi-ld/v1/entityOperations/upsert/?options=update",
    "method": "POST",
    "headers": {
        "fiware-service": "opcua_pcs",
        "fiware-servicepath": "/pcs",
        "Content-Type": "application/ld+json",
        "NGSILD-Tenant": "opcua_pcs",
        "NGSILD-Path": "/pcs"
    },
    "json": [
        {
            "@context": "http://ld-context/ld-context/datamodels.context-ngsild.jsonld",
            "id": "urn:ngsi-ld:PCS-STJ:DuctSystemHotAir",
            "type": "DuctSystemHotAir",
            "0HNB97CF901_LuMe_WL_TRFUnten_U": {
                "type": "Property",
                "value": {
                    "@type": "https://schema.org/Number",
                    "@value": 0.2824963629245758
                },
                "observedAt": "2024-09-26T15:45:59.444Z"
            }
        }
    ]
} | comp=IoTAgent
...
[... 747 lines cut ...]
 */
config.configRetrieval = false;
/**
 * Default API Key, to use with device that have been provisioned without a Configuration Group.
 */
config.defaultKey = 'iot';
/**
 * Default transport protocol when no transport is provisioned through the Device Provisioning API.
 */
config.defaultTransport = 'OPCUA';
/**
 * Flag indicating whether the node server will be executed in multi-core option (true) or it will be a
 * single-thread one (false).
 */
//config.multiCore = false;

module.exports = config;

```

#### Question

Can I somehow influence if the data is send as numeric or compound value format in the payload of the request of OPC UA iot agent to the northbound interface? I'd like to change something, so that the data is also send as numeric for the second data source. As the data schema has the same structure and there is no configuration option for compound vs. numeric value format in the agent configuration file, I am not sure how to influence the behavior of the agent.

Greetings and thank you in advance!

manfredipist, 2024-09-30:
Hi @IngMiad, you should try removing the schema url from type definitions 
eg: "https://schema.org/Number" -> "Number"
Could you give it a try? Let us know, best regards!

IngMiad, 2024-10-17:
Hello @manfredipist,

thank you very much this worked. 

Remark: In addition to removing the schema url from type definition, I had to remove the earlier devices entries from iot-agent mongo-db otherwise it still uses compound format.

Greetings!

closed 2024-10-17 by IngMiad